arduino-cmake/Arduino-CMake-NG #13: Arduino CMake cannot find build chain for new project on OSX

machadolab, 2018-09-17:
* Already have Arduino IDE 1.8.3 installed in /Applications
* git cloned Arduino-CMake-NG to /Users/machado/git/Arduino-CMake-NG
* Created ~/tmp/MyProject with the following:

CMakeLists.txt:
```
# Define CMake's minimum version (must-do) and the project's name and supported languages
cmake_minimum_required(VERSION 3.8)
project(Hello_World LANGUAGES C CXX ASM)

# Call a framework utility function, passing it information about the hardware board that will
# be used - This function returns a structure known only to the framework
get_board_id(board_id uno)

# Create an executable suitable for the Arduino firmware using CMake-style target-creation
add_arduino_executable(Hello_World ${board_id} helloWorld.cpp)
# Upload the created target through a connected Serial Port (Where your board is connected to)
#upload_arduino_target(Hello_World "${board_id}" [PORT])
```

helloWorld.cpp:
```
/*
  Blink
  Turns on an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the UNO, MEGA and ZERO 
  it is attached to digital pin 13, on MKR1000 on pin 6. LED_BUILTIN is set to
  the correct LED pin independent of which board is used.
  If you want to know what pin the on-board LED is connected to on your Arduino model, check
  the Technical Specs of your board  at https://www.arduino.cc/en/Main/Products
  
  This example code is in the public domain.

  modified 8 May 2014
  by Scott Fitzgerald
  
  modified 2 Sep 2016
  by Arturo Guadalupi
  
  modified 8 Sep 2016
  by Colby Newman
*/

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
```

And then tried to build:

```
[machado@mbp ~/tmp/MyProject]$ cmake -DCMAKE_TOOLCHAIN_FILE=/Users/machado/git/Arduino-CMake-NG/cmake/Arduino-Toolchain.cmake
-- The C compiler identification is AppleClang 9.1.0.9020039
-- The CXX compiler identification is AppleClang 9.1.0.9020039
-- The ASM compiler identification is AppleClang
-- Didn't find assembler
CMake Error at /Users/machado/git/Arduino-CMake-NG/cmake/Platform/System/AvrToolsFinder.cmake:11 (message):
  avr-objcopy program is required by the toolchain but can't be found
Call Stack (most recent call first):
  /Users/machado/git/Arduino-CMake-NG/cmake/Platform/System/BuildSystemInitializer.cmake:8 (find_tool_avr_objcopy)
  /Users/machado/git/Arduino-CMake-NG/cmake/Platform/System/BuildSystemInitializer.cmake:21 (find_required_platform_tools)
  /Users/machado/git/Arduino-CMake-NG/cmake/Platform/Arduino.cmake:34 (initialize_build_system)
  /Applications/CMake.app/Contents/share/cmake-3.12/Modules/CMakeSystemSpecificInformation.cmake:26 (include)
  CMakeLists.txt:3 (project)

CMake Error at CMakeLists.txt:3 (project):
  No CMAKE_ASM_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "/Users/machado/tmp/MyProject/CMakeFiles/CMakeOutput.log".
```

machadolab, 2018-09-17:
@taoyuan Your PR fixed me issue, thanks!!

MrPointer, 2018-09-18:
@machadolab Hi, a big thank you for contributing and bug reporting - We really appreciate that!
Can you take a look at the review I left in #14 which presumably fixes this issue? Maybe you could provide us with some insight on why does it happen, and more importantly - Why does the PR fixes it.

closed 2018-09-18 by MrPointer

machadolab, 2018-09-19:
@MrPointer the PR does indeed fix my issue. As to why, it simply seems like OSX has a different path than what was expected.

PR #14 (2018-09-18): Fix arduino sdk path finding issue on Mac
Fix  #13