arduino-cmake/Arduino-CMake-NG #15: Unable to use arduino libraries installed by library manager

machadolab, 2018-09-17:
I have tried both the **Built-in/Arduino Libraries** way and **User/3rd Party Libraries** way (as per the wiki documentation) to try and use a library that is installed with the Arduino IDE library manager. 

The first way does not work because it looks like Arduino-CMake-NG only searches the main Arduino libraries that come with the IDE software and not the user installed libraries that are installed into the **Sketchbook location** using the library manager. There also seems to be an issue with library naming. I have several popular arduino libs installed from Adafruit that have library names that do not conform to the ```arduino_compliant_library_name``` that Arduino-CMake-NG generates - for example Adafruit_NeoPixel and Adafruit_NeoMatrix are both converted to Adafruit_Neopixel and Adafruit_Neomatrix when the library directory is looked up.

The 3rd party library way did not work because the library uses arduino code and it looks like none of the arduino CFLAGs are passed to the building of the library. For example, I tried to define a user library for Adafruit_NeoPixel like so:

```
set(Adafruit_NeoPixel_Path /path/to/sketchbook/libraries/Adafruit_NeoPixel)
add_library(Adafruit_NeoPixel_lib STATIC ${Adafruit_NeoPixel_Path}/Adafruit_NeoPixel.cpp)
target_include_directories(Adafruit_NeoPixel_lib PUBLIC ${Adafruit_NeoPixel_Path})
link_arduino_library(Hello_World Adafruit_NeoPixel_lib ${board_id})
```

and get the build error:

```
In file included from /Users/machado/Dropbox/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp:34:0:
/Users/machado/Dropbox/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.h:25:23: fatal error: WProgram.h: No such file or directory
  #include <WProgram.h>
                       ^
compilation terminated.
make[3]: *** [CMakeFiles/Adafruit_NeoPixel_lib.dir/Users/machado/Dropbox/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp.obj] Error 1
make[2]: *** [CMakeFiles/Adafruit_NeoPixel_lib.dir/all] Error 2
make[1]: *** [CMakeFiles/Hello_World.dir/rule] Error 2
make: *** [Hello_World] Error 2
```

Because the define ARDUINO is not present and therefore it falls back to looking for WProgram.h.

MrPointer, 2018-09-18:
Hello @machadolab, and yet again welcome abroad!
You said that our framework doesn't search the **Sketchbook location**, and you're indeed correct.
I'll check this out, and see if there's anything I can extract from the SDK's metadata to add this location to our search paths.
However , from my understanding it's something that's tightly bound to **Arduino IDE**'s **Library Manager**, which our framework knows nothing about - Which would make it impossible to know a user's **Sketchbook location**.
If indeed that's the case, one would have no other choice but to manually pass this location to the framework - That is something that we can easily take care of.

Regarding the issue you've come across with 3rd party libraries - Could you please submit a new, separate issue for this? It would be easier for us to maintain.

machadolab, 2018-09-19:
So the issue is, since arduino libs need the same build flags as a 'built in' library, but the sketchbook location is not searched, there seems no way to build a project that uses these libs.

I suppose I could reverse engineer all the flags needed and simply treat it as a 'user library' but this seems way complicated since the library-manager installed libraries are basically exactly like the built in libs, just in a different location.

So I would suggest if you cant auto detect the sketchbook location, have a cmake parameter for them to specify the sketchbook location.

You should be able to easily reproduce this trouble by creating a very simple project using a Adafruit_NeoPixel example, for instance. You might also run into the same library naming issue I had too =)

MrPointer, 2018-09-20:
@machadolab I think we misunderstood each other partly 😃 
What I've meant is that you'd open a new issue for the bug regarding **Adafruit_NeoPixel**, since a failure to build this library (as well as all other "user" libraries) has nothing to do with the **Sketchbook Location**.
I'm saying this since I already have fixes for both, but the fixes resolve different issues, so it'd be nicer to have them managed as separate issues on GitHub as well.

However, I'll do this myself now to spare some time and just in case you still don't understand my intentions. Meanwhile, I'll also edit parts of this issue to attribute it solely to the **Sketchbook Location** "bug".

PR #20 (2018-09-20): Added support for Arduino IDE's Sketchbook
Arduino IDE's *Sketchbook* can now be used as an additional search path for libraries, especially 3rd party.

By default, the *Sketchbook* isn't automatically located, and to use it one should manually set the `ARDUINO_CMAKE_SKETCHBOOK_PATH` variable. However, this behavior can be changed by enabling the `AUTO_SET_SKETCHBOOK_PATH` option.

**Note:** *Sketchbook Location* is only valid after the **Arduino IDE** itself has run at least once, otherwise it won't create a `preferences` file which specifies *Sketchbook*'s location.

Fixes #15
closed 2018-09-20 by MrPointer