arduino-cmake/Arduino-CMake-NG #22: Missing ARDUINO_ARCH_<ARCH> define

machadolab, 2018-09-20:
Looks like the makefiles generated do not include the ARDUINO_ARCH_AVR define that is needed by some libs. I looked at the raw arduino ide build commands and they seem to have it.

```
/Applications/arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++   -I/Users/machado/Dropbox/Arduino/libraries/Adafruit_GFX -I/Users/machado/Dropbox/Arduino/libraries/Adafruit_GFX/Fonts -I/Users/machado/Dropbox/Arduino/libraries/Adafruit_NeoMatrix -I/Users/machado/Dropbox/Arduino/libraries/Adafruit_NeoPixel -I/Users/machado/Dropbox/Arduino/libraries/Adafruit_SSD1306 -I/Users/machado/Dropbox/Arduino/libraries/Arduino-SerialCommand -I/Users/machado/Dropbox/Arduino/libraries/ArduinoNunchuk -I/Users/machado/Dropbox/Arduino/libraries/EnableInterrupt -I/Users/machado/Dropbox/Arduino/libraries/EnableInterrupt/utility -I/Users/machado/Dropbox/Arduino/libraries/FastAPI_LED -I/Users/machado/Dropbox/Arduino/libraries/Keypad/src -I/Users/machado/Dropbox/Arduino/libraries/MFRC522/src -I/Users/machado/Dropbox/Arduino/libraries/MPU9250 -I/Users/machado/Dropbox/Arduino/libraries/MQTT/src -I/Users/machado/Dropbox/Arduino/libraries/MQTT/src/lib -I/Users/machado/Dropbox/Arduino/libraries/PinChangeInterrupt/src -I/Users/machado/Dropbox/Arduino/libraries/Skywriter -I/Users/machado/Dropbox/Arduino/libraries/TimerOne -I/Users/machado/Dropbox/Arduino/libraries/TimerOne/config -I/Users/machado/Dropbox/Arduino/libraries/Zumo32U4 -I/Users/machado/Dropbox/Arduino/libraries/grbl -I/Applications/arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/arduino.app/Contents/Java/hardware/arduino/avr/variants/standard -I/Applications/arduino.app/Contents/Java/hardware/arduino/avr/libraries/SPI/src -I/Applications/arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire/src -I/Applications/arduino.app/Contents/Java/hardware/arduino/avr/libraries/eeprom/src  -g   -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10803 "-DARDUINO_AVR_UNO " -o CMakeFiles/Skywriter.dir/Users/machado/Dropbox/Arduino/libraries/PinChangeInterrupt/src/PinChangeInterrupt.cpp.obj -c /Users/machado/Dropbox/Arduino/libraries/PinChangeInterrupt/src/PinChangeInterrupt.cpp
In file included from /Users/machado/Dropbox/Arduino/libraries/PinChangeInterrupt/src/PinChangeInterrupt.cpp:24:0:
/Users/machado/Dropbox/Arduino/libraries/PinChangeInterrupt/src/PinChangeInterrupt.h:33:2: error: #error This library can only be used with AVR
 #error This library can only be used with AVR
  ^
make[2]: *** [CMakeFiles/Skywriter.dir/Users/machado/Dropbox/Arduino/libraries/PinChangeInterrupt/src/PinChangeInterrupt.cpp.obj] Error 1
make[1]: *** [CMakeFiles/Skywriter.dir/all] Error 2
make: *** [all] Error 2
```

MrPointer, 2018-09-20:
@machadolab Thanks for reporting yet again!
Could you please attach the output of **CMakeFiles/CMakeOutput.log** which is located under your build directory?

MrPointer, 2018-09-22:
@machadolab Just out of curiosity - Does the problem persists after the last bug-fixes? 
I'd like to check it against the mainstream **develop** branch if you could.

MrPointer, 2018-09-22:
Oh, really glad to hear that - Definitely didn't expect that 🤣