arduino-cmake/Arduino-CMake-NG #23: Build tries to build all .c files of a library which might not have all the dependent headers available

machadolab, 2018-09-20:
When using the find_arduino_library/link_arduino_library to include a library from my arduino sketchbook location, it looks like the build is trying to grab all .c files that are in the library. Sometimes libraries include tools or example code that might depend on headers that are not on the system, leading to a failed build, even though the code/headers for the library itself are just fine.

For example, I am using the Adafruit_GFX library that includes a font utility that depends on some freetype libs. The actual Adafruit_GFX lib does not depend on the font utility, its just there in case you want to use it, but since the utility cannot be built on my system (freetype headers unavailable) it causes my project build to fail:

```
[ 34%] Building C object CMakeFiles/Skywriter.dir/Users/machado/Dropbox/Arduino/libraries/Adafruit_GFX/fontconvert/fontconvert.c.obj
/Applications/arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc  -I/Users/machado/Dropbox/Arduino/libraries/Adafruit_GFX -I/Users/machado/Dropbox/Arduino/libraries/Adafruit_GFX/Fonts -I/Users/machado/Dropbox/Arduino/libraries/Adafruit_NeoMatrix -I/Users/machado/Dropbox/Arduino/libraries/Adafruit_NeoPixel -I/Users/machado/Dropbox/Arduino/libraries/Adafruit_SSD1306 -I/Users/machado/Dropbox/Arduino/libraries/Arduino-SerialCommand -I/Users/machado/Dropbox/Arduino/libraries/ArduinoNunchuk -I/Users/machado/Dropbox/Arduino/libraries/EnableInterrupt -I/Users/machado/Dropbox/Arduino/libraries/EnableInterrupt/utility -I/Users/machado/Dropbox/Arduino/libraries/FastAPI_LED -I/Users/machado/Dropbox/Arduino/libraries/Keypad/src -I/Users/machado/Dropbox/Arduino/libraries/MFRC522/src -I/Users/machado/Dropbox/Arduino/libraries/MPU9250 -I/Users/machado/Dropbox/Arduino/libraries/MQTT/src -I/Users/machado/Dropbox/Arduino/libraries/MQTT/src/lib -I/Users/machado/Dropbox/Arduino/libraries/PinChangeInterrupt/src -I/Users/machado/Dropbox/Arduino/libraries/Skywriter -I/Users/machado/Dropbox/Arduino/libraries/TimerOne -I/Users/machado/Dropbox/Arduino/libraries/TimerOne/config -I/Users/machado/Dropbox/Arduino/libraries/Zumo32U4 -I/Users/machado/Dropbox/Arduino/libraries/grbl -I/Applications/arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/arduino.app/Contents/Java/hardware/arduino/avr/variants/standard -I/Applications/arduino.app/Contents/Java/hardware/arduino/avr/libraries/SPI/src -I/Applications/arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire/src -I/Applications/arduino.app/Contents/Java/hardware/arduino/avr/libraries/eeprom/src  -g   -o CMakeFiles/Skywriter.dir/Users/machado/Dropbox/Arduino/libraries/Adafruit_GFX/fontconvert/fontconvert.c.obj   -c /Users/machado/Dropbox/Arduino/libraries/Adafruit_GFX/fontconvert/fontconvert.c
/Users/machado/Dropbox/Arduino/libraries/Adafruit_GFX/fontconvert/fontconvert.c:23:22: fatal error: ft2build.h: No such file or directory
 #include <ft2build.h>
                      ^
compilation terminated.
make[2]: *** [CMakeFiles/Skywriter.dir/Users/machado/Dropbox/Arduino/libraries/Adafruit_GFX/fontconvert/fontconvert.c.obj] Error 1
make[1]: *** [CMakeFiles/Skywriter.dir/all] Error 2
make: *** [all] Error 2
```

MrPointer, 2018-09-20:
@machadolab Well here's the thing:
As part of the support for 3rd Party libraries and the ability to treat them as "first-class" **Arduino Libraries**, I had to make the framework search for sources (and headers) recursively under the library's root directory, unlike the older version which searched only under the 'src' directory (That's how built-in libs are structured).

Why am I telling you this? In order to truly find only sources that are relevant to the library itself, I had to exclude any sources not matching a certain pattern. In the case of the **Adafruit_NeoPixel** library which I've tested against, it's only the **examples** directory that should be excluded.
However, with this new type of library, it seems that **fonts** should be excluded too, and that's fine, but it leads me to think that there might be countless directory named which should be excluded, and the framework could never know that in advance.
Besides, excluding a directory to support one library could break the build for another, if the exclude pattern will name these directories explicitly.

It'll take some time to solve this - Any help or ideas are welcomed.

MrPointer, 2018-09-20:
Nope, absolutely no idea.
It'll be impossible for me to dig into their code right now to extract it, too much effort and time to spend...

machadolab, 2018-09-20:
Found this that might help: https://github.com/arduino/Arduino/wiki/Build-Process

What was the motivation for during the recursive code search in the first place?

MrPointer, 2018-09-20:
Thanks, however, there's little I can learn from these specifications regarding our issue.
In fact, the [Library Directory Structure/Layout Specifications](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#layout-of-folders-and-files) specify that we're doing everything right in the framework.
Maybe, just maybe, the **Arduino IDE** reads all of your project's/sketches include files and only adds what it finds besides the hidden ones it always adds when building, such as `Arduino.h`... I hope that's not the case as it would make the framework extremely messy in that prospect.

machadolab, 2018-09-20:
Excellent!

PR #25 (2018-09-21): Modified framework's approach to libraries directory layout
From #23, it appears that the **Arduino-CMake-NG** framework hadn't had the ability to correctly manage **Arduino Libraries**'s sources since some of them were added when not required, some not added at all.

This PR gives the framework a consistent approach to **Arduino Libraries**, where the following rules apply:
* All sources under the **src** sub-directory and _**all of its sub-directories**_ are included in the library target's sources, if the **src** sub-directory exists
* If it doesn't, only the sources located directly under  [... cut]
closed 2018-09-21 by MrPointer