arduino-cmake/Arduino-CMake-NG #28: Not finding arduino-installed sketchbook libs that use 1.0 library format

machadolab, 2018-09-21:
It looks like the new code finds the 1.5 libs just fine. Thanks!!

But I've tried 3 different libs that use the old 1.0 format (no src/ directory) and none of them can be found. Here is an example using the EnableInterrupt library:

(as an aside, there seems be a problem again finding the arduino SDK on OSX. I had to set the system environment variable ARDUINO_SDK_PATH in order for it to work (but there is still funny log output with NOTFOUND in the output). (I'll open another bug for this).

CMakeLists.txt library entry:
```
find_arduino_library(EnableInterrupt EnableInterrupt ${board_id})
link_arduino_library(Hello_World EnableInterrupt ${board_id})
```

Result:

```
-- Arduino SDK version 1.8.3: ARDUINO_SDK_PATH-NOTFOUND
-- Determined Platform Header: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h
-- Arduino SDK version 1.8.3: ARDUINO_SDK_PATH-NOTFOUND
-- Determined Platform Header: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h
-- Arduino SDK version 1.8.3: ARDUINO_SDK_PATH-NOTFOUND
-- Determined Platform Header: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h
CMake Error at cmake/Platform/Targets/ArduinoLibraryTarget.cmake:136 (message):
  Couldn't find any source files for the Enableinterrupt library
Call Stack (most recent call first):
  CMakeLists.txt:28 (find_arduino_library)

CMake Error at cmake/Platform/Targets/ArduinoLibraryTarget.cmake:162 (message):
  Library target doesn't exist - It must be created first!
Call Stack (most recent call first):
  CMakeLists.txt:29 (link_arduino_library)

-- Configuring incomplete, errors occurred!
See also "/Users/machado/tmp/MyProject/CMakeFiles/CMakeOutput.log".
```

MrPointer, 2018-09-22:
@machadolab I was very skeptical about this being a "bug" when I first read your description, mostly because all the Adafruit examples (that are now also part of the framework's official examples) are compliant with the 1.0 standard, not the 1.5!
Then I took a look at the [EnableInterrupt](https://github.com/GreyGnome/EnableInterrupt) library, and noticed something a little bit tricky which I haven't thought of - It's a header only library! Am I right?
If so, the framework indeed doesn't support these kind of libraries right now, but I'll fix that ASAP, it's an easy one!

> Disclaimer: I don't know anything about Arduino development, I only manage this repo because of my CMake knowledge and desire to gain some more experience in it, as well as maintaining an open source project. So the next time you describe a problem that you have, especially if it regards some 3rd party libraries, please instruct me where can I find the library (a link would suffice) - That's the minimum I request that you could help me with 😃

machadolab, 2018-09-22:
Ah I didn’t notice that about EnableInterrupt, but I have seen that in the past so ya it should support it. The other library that I had a problem with is Skywriter (https://github.com/pimoroni/skywriter-hat/tree/master/arduino/Skywriter) which I did not install via the library manager so I wasn’t sure if that had something to do with it. This lib appears to conform to 1.0 standard and has both .cpp and .h files, so sounds like a different problem. If you can see why this lib isn’t working, I’d much appreciate it!

Thanks for all the hard work - this project is great! For the record my motivation to get this working well is that I use CLion IDE which is way nicer, as a professional software developer, than the arduino IDE and is 100% cmake based. So far it’s been working well with this project!

MrPointer, 2018-09-22:
Good thing I develop this framework in CLion myself 😆 Absolutely love this IDE! I'm considering developing a plugin to this framework once it'll get to 1.0, but we'll see about that lol

Back to business - I had a look at the **Skywriter** library and it appears as a classical example of a user library which tricks you to think it's an Arduino library.
Notice that it doesn't have the **library.properties** file that our framework is looking for when searching for Arduino libraries, so it can't extract basic metadata from this library.

There several ways to solve this issue probably, and I honestly don't know which is best. I'll list them when I'll have time to attend this issue, and we'll decide together 😃

machadolab, 2018-09-22:
I had the same thought about a CLion plugin based on this work! :)

As for the library issue - I was re-reading the spec and based on this section https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#library-metadata it sounds like library.properties file was added as part of the 1.5 spec. So I would expect that not many, if any, 1.0 libs would have this file, so I'd suggest just the existence of the library directory as the signal that the library exists.

MrPointer, 2018-09-22:
You're right, I did miss that section about the file being added only in the 1.5 standard.
However, we might still require this metadata. What is this metadata anyways?
Currently, the framework only reads the **architecture** key, which lists all hardware archs supported by the library, or '*' if the library is arch-agnostic.

One way to tackle our problem is by simply assuming that a library that doesn't list the **library.properties** file is in fact arch-agnostic, and hope that this is indeed the case.
The only problem with this approach is that Arduino might introduce additional important metadata in this file, which will require the framework to read this file no matter what. I can live with that, dynamically updating the framework's approach and code when the time comes, if ever.

The 2nd option is to simply treat it as a user library that the user should manually provide.

@machadolab What option do you like better? Or is there anything else you'd like to suggest?

machadolab, 2018-09-22:
I would assume the arch is *. If there are problems, the lib maintainer should update it to 1.5 with the correct arch in the metadata. The 2nd option might not work well, because libs often require the arduino build CFLAGS, and I am guessing those wouldnt be automatically added to user libs.

MrPointer, 2018-09-22:
@machadolab You're absolutely right, user libraries don't get any special Arduino flags and that options is definitely worse!
With that in mind, I'll work on this behavior soon!

MrPointer, 2018-09-26:
@machadolab Checkout the new **develop** branch - It includes a fix to this issue 😃

PR #39 (2018-09-26): Added support for libraries that don't have a properties file
This feature required the core find-lib process to change as such: 
> Instead of searching for the properties file to determine whether a library is found or not, its' root directory is searched instead.

Due to latest refactoring changes, the `add_arduino_library` also handles finding the properties file itself, so all the `find_arduino_library` function does now is gather the required sources and headers.

An example of a library without a properties file has also been added, testing against the **Skywriter** library.

This PR als [... cut]
closed 2018-09-26 by MrPointer