arduino-cmake/Arduino-CMake-NG #29: Arduino SDK not automatically detected on OSX

machadolab, 2018-09-21:
```
CMake Error at cmake/Platform/Other/ArduinoSDKSeeker.cmake:32 (message):
  Couldn't find Arduino SDK path - Is it in a non-standard location?

  If so, please set the ARDUINO_SDK_PATH CMake-Variable
Call Stack (most recent call first):
  cmake/Arduino-Toolchain.cmake:66 (find_arduino_sdk)
  /Applications/CMake.app/Contents/share/cmake-3.12/Modules/CMakeDetermineSystem.cmake:91 (include)
  CMakeLists.txt:3 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/machado/tmp/MyProject/CMakeFiles/CMakeOutput.log".
```

When I set the system environment variable ```ARDUINO_SDK_PATH``` to ```/Applications/Arduino.app/Contents/Java``` it is able to finish cmake, but there is some strange output:

```
-- Arduino SDK version 1.8.3: ARDUINO_SDK_PATH-NOTFOUND
-- Determined Platform Header: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h
-- Arduino SDK version 1.8.3: ARDUINO_SDK_PATH-NOTFOUND
-- Determined Platform Header: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h
-- Arduino SDK version 1.8.3: ARDUINO_SDK_PATH-NOTFOUND
-- Determined Platform Header: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/machado/tmp/MyProject
```

taoyuan, 2018-09-22:
The Arduino SDK on OSX should be in subdirectory `Arduino.app/Contents/Java` of parent folder `Applications`.

I have fixed it [here](https://github.com/taoyuan/Arduino-CMake-NG/commit/d0c9ab80ecf60a88c69861a10dd2052c03aae703). If the fix is acceptable, I can make a PR.

The strange output caused by the environment variable `ARDUINO_SDK_PATH ` really needs to review the code.

taoyuan, 2018-09-22:
The Arduino SDK on OSX should be in subdirectory `Arduino.app/Contents/Java` of parent folder `Applications`.

I have create a PR #30 to fix this.

The strange output caused by the environment variable `ARDUINO_SDK_PATH ` really needs to review the code.

MrPointer, 2018-09-22:
@machadolab @taoyuan I'll have to check that output, although I have a theory on why it happens - the new _Find-SDK module_ looks for the **lib/version.txt** file and stores the found path in a variable named `ARDUINO_SDK_PATH`. If it hasn't found anything, this variable will be set to `ARDUINO_SDK_PATH-NOTFOUND` (this how all CMake's `find_xxx` functions work).
However, even when the variable is set manually on the command-line, the _Find module_ **overrides** this value, according to CMake's variable scoping rules, and that's how you get that strange output.
The reason I chose to set it to a variable with the same name is to avoid declaring more variables which don't actually have any meaning, but apparently now they do...

MrPointer, 2018-09-22:
Ummm this kind of bothers me because I did NOTHING since the last comment, just shared my thoughts... I still need to investigate this one further, but I'll open a separate issue for this