arduino-cmake/Arduino-CMake-NG #34: Support for external boards

ooxi, 2018-09-25:
I tried your CMake toolchain and it works really nice for AVR based systems! However (just as you told in the documentation) it does not work for other boards.

Is there a roadmap on supporting other boards? My use case is the Adafruit nRF52

MrPointer, 2018-09-25:
First of all great to hear that it works for you!
There's an upcoming release, 0.6, which fixes numerous bugs and adds some more features, so the "base" framework is even getting better 😃 

As to your question - There's a partial road map that you can find [here](https://github.com/arduino-cmake/Arduino-CMake-NG/projects/2) - It doesn't specify too much now because I don't have enough time to work on it as well resolving bugs in the already-existing code.
Besides, I'd like to first complete support for [Arduino's fundamentals](https://github.com/arduino-cmake/Arduino-CMake-NG/projects/3) - Any platform-generic solution will be based on this anyways!

ooxi, 2018-09-25:
Thank you for the fast response and keep up the good work!

closed 2018-09-25 by ooxi