arduino-cmake/Arduino-CMake-NG #43: Problem with linking Wire lib

andrzejo, 2018-10-05:
I have build problem with project using Wire lib:

CMakeLists.txt
```
cmake_minimum_required(VERSION 3.8)

set(CMAKE_TOOLCHAIN_FILE ${CMAKE_SOURCE_DIR}/cmake/Arduino-Toolchain.cmake)

project(lcd LANGUAGES C CXX ASM)

get_board_id(board_id mega atmega2560)

add_arduino_executable(lcd ${board_id} src/lcd.cpp)

find_arduino_library(crystal LiquidCrystal_I2C ${board_id} 3RD_PARTY)
link_arduino_library(lcd crystal ${board_id})
```
src/lcd.cpp
```
#include <Arduino.h>

#include <LiquidCrystal_I2C.h>

#define LCD_ADDRESS     0x27
#define LCD_COLUMNS     20
#define LCD_ROWS        4

LiquidCrystal_I2C lcd(LCD_ADDRESS, LCD_COLUMNS, LCD_ROWS);

void setup() {
    Serial.begin(9600);
    lcd.begin(LCD_COLUMNS, LCD_ROWS);
    lcd.clear();
}

void loop() {
}
```

Library **LiquidCrystal_I2C** use **Wire** library and it should be linked to executable, but there is an linking error: 
+ make
```
Scanning dependencies of target mega_atmega2560_core_lib
[  3%] Building C object CMakeFiles/mega_atmega2560_core_lib.dir/opt/arduino/hardware/arduino/avr/cores/arduino/WInterrupts.c.obj
[  6%] Building C object CMakeFiles/mega_atmega2560_core_lib.dir/opt/arduino/hardware/arduino/avr/cores/arduino/hooks.c.obj
[  9%] Building C object CMakeFiles/mega_atmega2560_core_lib.dir/opt/arduino/hardware/arduino/avr/cores/arduino/wiring.c.obj
[ 12%] Building C object CMakeFiles/mega_atmega2560_core_lib.dir/opt/arduino/hardware/arduino/avr/cores/arduino/wiring_analog.c.obj
[ 15%] Building C object CMakeFiles/mega_atmega2560_core_lib.dir/opt/arduino/hardware/arduino/avr/cores/arduino/wiring_digital.c.obj
[ 18%] Building C object CMakeFiles/mega_atmega2560_core_lib.dir/opt/arduino/hardware/arduino/avr/cores/arduino/wiring_pulse.c.obj
[ 21%] Building C object CMakeFiles/mega_atmega2560_core_lib.dir/opt/arduino/hardware/arduino/avr/cores/arduino/wiring_shift.c.obj
[ 25%] Building CXX object CMakeFiles/mega_atmega2560_core_lib.dir/opt/arduino/hardware/arduino/avr/cores/arduino/CDC.cpp.obj
[ 28%] Building CXX object CMakeFiles/mega_atmega2560_core_lib.dir/opt/arduino/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp.obj
[ 31%] Building CXX object CMakeFiles/mega_atmega2560_core_lib.dir/opt/arduino/hardware/arduino/avr/cores/arduino/HardwareSerial0.cpp.obj
[ 34%] Building CXX object CMakeFiles/mega_atmega2560_core_lib.dir/opt/arduino/hardware/arduino/avr/cores/arduino/HardwareSerial1.cpp.obj
[ 37%] Building CXX object CMakeFiles/mega_atmega2560_core_lib.dir/opt/arduino/hardware/arduino/avr/cores/arduino/HardwareSerial2.cpp.obj
[ 40%] Building CXX object CMakeFiles/mega_atmega2560_core_lib.dir/opt/arduino/hardware/arduino/avr/cores/arduino/HardwareSerial3.cpp.obj
[ 43%] Building CXX object CMakeFiles/mega_atmega2560_core_lib.dir/opt/arduino/hardware/arduino/avr/cores/arduino/IPAddress.cpp.obj
[ 46%] Building CXX object CMakeFiles/mega_atmega2560_core_lib.dir/opt/arduino/hardware/arduino/avr/cores/arduino/PluggableUSB.cpp.obj
[ 50%] Building CXX object CMakeFiles/mega_atmega2560_core_lib.dir/opt/arduino/hardware/arduino/avr/cores/arduino/Print.cpp.obj
[ 53%] Building CXX object CMakeFiles/mega_atmega2560_core_lib.dir/opt/arduino/hardware/arduino/avr/cores/arduino/Stream.cpp.obj
[ 56%] Building CXX object CMakeFiles/mega_atmega2560_core_lib.dir/opt/arduino/hardware/arduino/avr/cores/arduino/Tone.cpp.obj
[ 59%] Building CXX object CMakeFiles/mega_atmega2560_core_lib.dir/opt/arduino/hardware/arduino/avr/cores/arduino/USBCore.cpp.obj
[ 62%] Building CXX object CMakeFiles/mega_atmega2560_core_lib.dir/opt/arduino/hardware/arduino/avr/cores/arduino/WMath.cpp.obj
[ 65%] Building CXX object CMakeFiles/mega_atmega2560_core_lib.dir/opt/arduino/hardware/arduino/avr/cores/arduino/WString.cpp.obj
[ 68%] Building CXX object CMakeFiles/mega_atmega2560_core_lib.dir/opt/arduino/hardware/arduino/avr/cores/arduino/abi.cpp.obj
[ 71%] Building CXX object CMakeFiles/mega_atmega2560_core_lib.dir/opt/arduino/hardware/arduino/avr/cores/arduino/main.cpp.obj
[ 75%] Building CXX object CMakeFiles/mega_atmega2560_core_lib.dir/opt/arduino/hardware/arduino/avr/cores/arduino/new.cpp.obj
[ 78%] Building ASM object CMakeFiles/mega_atmega2560_core_lib.dir/opt/arduino/hardware/arduino/avr/cores/arduino/wiring_pulse.S.obj
[ 81%] Linking CXX static library libmega_atmega2560_core_lib.a
[ 81%] Built target mega_atmega2560_core_lib
Scanning dependencies of target Wire
[ 84%] Building CXX object CMakeFiles/Wire.dir/opt/arduino/hardware/arduino/avr/libraries/Wire/src/Wire.cpp.obj
[ 87%] Linking CXX static library libWire.a
[ 87%] Built target Wire
Scanning dependencies of target crystal
[ 90%] Building CXX object CMakeFiles/crystal.dir/libraries/LiquidCrystal_I2C/LiquidCrystal_I2C.cpp.obj
[ 93%] Linking CXX static library libcrystal.a
[ 93%] Built target crystal
Scanning dependencies of target lcd
[ 96%] Building CXX object CMakeFiles/lcd.dir/src/lcd.cpp.obj
[100%] Linking CXX executable lcd.elf
/tmp/ccNGShVI.ltrans0.ltrans.o: In function `write':
/opt/arduino/hardware/arduino/avr/libraries/Wire/src/Wire.cpp:228: undefined reference to `twi_transmit'
/opt/arduino/hardware/arduino/avr/libraries/Wire/src/Wire.cpp:210: undefined reference to `twi_transmit'
/tmp/ccNGShVI.ltrans0.ltrans.o: In function `endTransmission':
/opt/arduino/hardware/arduino/avr/libraries/Wire/src/Wire.cpp:173: undefined reference to `twi_writeTo'
collect2: error: ld returned 1 exit status
CMakeFiles/lcd.dir/build.make:98: recipe for target 'lcd.elf' failed
make[2]: *** [lcd.elf] Error 1
CMakeFiles/Makefile2:144: recipe for target 'CMakeFiles/lcd.dir/all' failed
make[1]: *** [CMakeFiles/lcd.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```
All files are available at [https://github.com/andrzejo/cmake-test](url)

MrPointer, 2018-10-05:
@andrzejo Hello and welcome to the project!
Can you fix the url pointing to your files please? It's currently invalid.

andrzejo, 2018-10-05:
Sure! It's valid now.

MrPointer, 2018-10-05:
Thanks.
One more thing - Since I don't know much about Arduino development (Ironic, I know) I need some help finding the exact libraries you're working with. So I'd like you to also point me to the repo/url of the **LiquidCrystal_I2C ** library you're using 😃

andrzejo, 2018-10-05:
I prepare console build script (required bash & linux), when you clone my test repo and run `./build.sh` script it should download required libraries (and cmake) and try to build application. If you running linux the build script should be helpful. 

[view build.sh](https://github.com/andrzejo/cmake-test/blob/master/build.sh)
 
Other way is to clone lib into libraries directory:
`git clone https://github.com/marcoschwartz/LiquidCrystal_I2C.git`

MrPointer, 2018-10-05:
Thanks again, I'll check this out in the following days

theunkn0wn1, 2018-10-05:
I am also having issues with using the `Wire` arduino library, Arduino-cmake is refusing to load it, which may be relevant to the OP's issue

relevant portion of the `CMakeLists.txt`:
```cmake
cmake_minimum_required(VERSION 3.8)

project(Robotics LANGUAGES C CXX ASM)
get_board_id(board_id mega atmega2560)
#set(SD_RECURSE True)
#set(Wire_RECURSE True)
# define our main executable
add_arduino_executable(Robotics
        ${board_id}
        i2c_playground.cpp
        )

find_arduino_library(WireLib Wire ${board_id})
```

cmake output
```
D:\programfiles\Jetbrains\apps\CLion\ch-0\182.4505.18\bin\cmake\win\bin\cmake.exe -DCMAKE_BUILD_TYPE=Debug -DCMAKE_TOOLCHAIN_FILE=Arduino-CMake-NG/cmake/Arduino-Toolchain.cmake -G "CodeBlocks - MinGW Makefiles" D:\Clion\skool\cs1b\robotics
-- Arduino SDK version 1.8.7: D:/Program Files (x86)/Arduino
-- Determined Platform Header: D:/Program Files (x86)/Arduino/hardware/arduino/avr/cores/arduino/Arduino.h
-- Arduino SDK version 1.8.7: D:/Program Files (x86)/Arduino
-- Determined Platform Header: D:/Program Files (x86)/Arduino/hardware/arduino/avr/cores/arduino/Arduino.h
-- Arduino SDK version 1.8.7: D:/Program Files (x86)/Arduino
-- Determined Platform Header: D:/Program Files (x86)/Arduino/hardware/arduino/avr/cores/arduino/Arduino.h
CMake Error at Arduino-CMake-NG/cmake/Platform/Targets/ArduinoLibraryTarget.cmake:125 (message):
  Couldn't find library named Wire
Call Stack (most recent call first):
  CMakeLists.txt:13 (find_arduino_library)

CMake Error at CMakeLists.txt:14 (link_arduino_library):
  link_arduino_library Function invoked with incorrect arguments for function
  named: link_arduino_library
```

MrPointer, 2018-10-06:
@theunkn0wn1 Hi there - Your issue seems a completely different one, could you please open a separate issue for this?

MrPointer, 2018-10-06:
@andrzejo I couldn't reproduce your issue but found another one instead, I'll integrate the fix soon.
However, I suspect you're not using the latest version of the framework.
After the fix will be integrated, please checkout the updated version of the **master** branch and let me know if the bug is indeed fixed!

PR #46 (2018-10-06): Fixed bug where platform libraries weren't using all of their sources
Only the root directory of a platform library were searched for sources, even if a utility sub-directory exists.
This is fixed now as the same search algorithm used for Arduino Libraries is applied here as well.

Also fixed a bug where libraries that support a single architecture which happens to be the platform architecture weren't supported - This case wasn't covered in the cmake code.

Fixes #43.
closed 2018-10-06 by MrPointer