arduino-cmake/Arduino-CMake-NG #49: CLion integration not working (include error)

jenswet, 2018-10-16:
Hello,

I tried to create a new CLion project and copied the 0.6 release at my project.

This is my CMakeLists.txt:
```
cmake_minimum_required(VERSION 3.12)
set(CMAKE_TOOLCHAIN_FILE ${CMAKE_SOURCE_DIR}/cmake/Arduino-Toolchain.cmake)
project(Arduino LANGUAGES C)

get_board_id(board_id mega atmega2560)
add_arduino_executable(${PROJECT_NAME} ${board_id} main.c)
upload_arduino_target(${PROJECT_NAME} "${board_id}" COM3)
```

My folder structure is:
Arduino (CLion project root)
- cmake (contains the whole 0.6 release)
- CMakeLists.txt
- main.c

Unfortunately I get this error.
```
/Applications/CLion.app/Contents/bin/cmake/mac/bin/cmake -DCMAKE_BUILD_TYPE=Debug -G "CodeBlocks - Unix Makefiles" /Users/jens/Downloads/Arduino
CMake Error at cmake/Arduino-Toolchain.cmake:1 (include):
  include could not find load file:

    Platform/Other/ArduinoSDKSeeker.cmake
Call Stack (most recent call first):
  /Applications/CLion.app/Contents/bin/cmake/mac/share/cmake-3.12/Modules/CMakeDetermineSystem.cmake:94 (include)
  CMakeLists.txt:3 (project)

CMake Error at cmake/Arduino-Toolchain.cmake:66 (find_arduino_sdk):
  Unknown CMake command "find_arduino_sdk".
Call Stack (most recent call first):
  /Applications/CLion.app/Contents/bin/cmake/mac/share/cmake-3.12/Modules/CMakeDetermineSystem.cmake:94 (include)
  CMakeLists.txt:3 (project)

CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!

[Finished]
```

I couldn't find any solution on github. I already tried a cmake command line parameter instead of the `set(CMAKE_TOOLCHAIN_FILE ${CMAKE_SOURCE_DIR}/cmake/Arduino-Toolchain.cmake)`. This didn't help. I also tried different relative and absolute paths without success.

Can anyone help me?

Thank you :)

MrPointer, 2018-10-19:
@jenswet Hello there, I'm sorry for being late and answering only now, I'm currently on an extremely busy schedule.
It's a strange issue you got there since I'm developing in **CLion** myself, as well as many others involved in this project and it works just fine for us.

I'd start with replacing the `set(CMAKE_TOOLCHAIN_FILE ${CMAKE_SOURCE_DIR}/cmake/Arduino-Toolchain.cmake)` line with a command-line variable (which you can set in **Settings**->**CMake**).
I know you already tried that, but my guess is that something wasn't configured right:
It's probably worth mentioning that you should pass **an absolute path** to the command variable, otherwise it won't work!
At last, remember to _reset_ cache after the change and not just reload it!

Please let me know if that solved your problem, and attach the output if it's not.

jenswet, 2018-10-23:
Hello @MrPointer,
thanks for your answer.

Unfortunately I still can't get it to work:

```
/Applications/CLion.app/Contents/bin/cmake/mac/bin/cmake -DCMAKE_BUILD_TYPE=Debug -DCMAKE_TOOLCHAIN_FILE=/Users/jens/Downloads/Arduino/cmake/Arduino-Toolchain.cmake -G "CodeBlocks - Unix Makefiles" /Users/jens/Downloads/Arduino
CMake Error at cmake/Arduino-Toolchain.cmake:1 (include):
  include could not find load file:

    Platform/Other/ArduinoSDKSeeker.cmake
Call Stack (most recent call first):
  /Applications/CLion.app/Contents/bin/cmake/mac/share/cmake-3.12/Modules/CMakeDetermineSystem.cmake:94 (include)
  CMakeLists.txt:3 (project)

CMake Error at cmake/Arduino-Toolchain.cmake:66 (find_arduino_sdk):
  Unknown CMake command "find_arduino_sdk".
Call Stack (most recent call first):
  /Applications/CLion.app/Contents/bin/cmake/mac/share/cmake-3.12/Modules/CMakeDetermineSystem.cmake:94 (include)
  CMakeLists.txt:3 (project)

CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!

[Finished]
```

My new CMakeLists.txt:
```
cmake_minimum_required(VERSION 3.8.2)
project(Arduino)

get_board_id(board_id mega atmega2560)
add_arduino_executable(${PROJECT_NAME} ${board_id} main.c)
upload_arduino_target(${PROJECT_NAME} "${board_id}" COM3)
```

Obviously it still doesn't like my installation.

Which operating system are you using? I'm on macOs 10.13.6.

MrPointer, 2018-10-23:
@jenswet May I ask if this path `/Users/jens/Downloads/Arduino` is the path to your **CMakeLists.txt** file?

As to your question - I'm using Windows 10 but it doesn't really matter as it seems it doesn't work for you at all, not even when using the command-line, which is really strange since we've got this covered in our CI builds - We build for OS X, Linux and Windows.

jenswet, 2018-10-23:
Yes this is the absolute path to CmakeLists.
Can I provide any Information for debugging?

MrPointer, 2018-10-23:
Did you install the project and all of its dependencies just as listed in the [installation notes](https://github.com/arduino-cmake/Arduino-CMake-NG/wiki/Installation)?

jenswet, 2018-10-25:
I used the CLion CMake bundled before.
Now I installed CMake as additional dependency and configured it as Toolchain but it does not change the situation.
Arduino IDE is installed as usual.

MrPointer, 2018-12-01:
@jenswet I know I'm extremely late, but is it still relevant?

If so, could you please attach the output of your project's directory listing? I mean to execute `ls -lR` from the terminal.

MrPointer, 2019-03-08:
Closing due to inactivity. Feel free to reopen if needed.

closed 2019-03-08 by MrPointer