arduino-cmake/Arduino-CMake-NG #50: Document board discovery, `get_board_id` parameters

Yona-Appletree, 2018-10-22:
I can't figure out, and don't see in docs, information about how to figure out what the correct identifiers are for `get_board_id`, or how to list the available parameters.

It would be really useful to have a function, like the `print_board_list()` function in the old queezythegreat repo.

MrPointer, 2018-10-22:
@Yona-Appletree You're absolutely right - It should be documented.
I wonder though what value does the `print_board_list` provides? Is it really that useful?

Yona-Appletree, 2018-10-22:
Perhaps -- maybe you have a better solution. My situation is that I have boards listed in the Arduino IDE that I want to use, and of course all I see is the name in the menu. How do I select the STM32F103C?

For what it's worth, I used this bit of shell to list all the valid identifiers: `find ~/Library/Arduino15 -name boards.txt | xargs egrep . | sed -E 's/#.*//' | egrep -o '([^/]+/)+boards.txt:[^\.]+' | sort | uniq | grep --color ':.*'`

However, when I actually used one of those, in my case `genericSTM32F103C`, I got an error stating that it was invalid. Presumably I need to tell arduino-cmake about that folder? In any case, some documentation and examples would be awesome. Thanks!

MrPointer, 2018-10-26:
If I understand correctly, `STM32F103C` isn't supported by the basic Arduino SDK, since I can't find it in the typical **board.txt** file.
If that's true, it means that it's introduced by some other platform, a 3rd party platform if you'd like, and that is currently **not supported**. 
It is however one of the project's main goals and it won't reach version 1.0 until this is fully implemented.

As to the original question, yes, I should document the entire `board_id` concept/API.