arduino-cmake/Arduino-CMake-NG #52: Using additional board manager json files

toshi38, 2018-10-24:
I think I have a similar issue to #50 I'm a bit new to the arduino ecosystem so not sure _exactly_ how this is supposed to work.  That said I have a link `https://dl.espressif.com/dl/package_esp32_index.json` that I add to the `Additional Boards Manager URLs` field in Arduino IDE.  This (based on a little bit of poking around) links eventually to `https://github.com/espressif/arduino-esp32/blob/master/boards.txt` which of course has the board that I need to find with `get_board_id`.

So the question is how am I supposed to access this?
- Should I copy this into my `boards.txt`?
- Install via IDE (doesn't seem to get found by cmake)
- Other?

And follow up question, should we consider adding another function like `add_additional_board_manager` that downloads this file (and w/e else is needed to support) such that `get_board_id` would then _just work_?

toshi38, 2018-10-24:
Update:

So I followed the instructions here:
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/debian_ubuntu.md

But the `get_board_id` still doesn't work, I did a bit of digging and it seems like `initialize_arduino_platform` function is only looking in the `arduino-1.8.7/hardware/arduino/avr` path but not searching the `arduino-1.8.7/hardware/espressif/esp32` path.

Of course we can force the paths like so:
```
#Setup paths to use esp32
set(ARDUINO_SDK_PATH "/usr/share/arduino-1.8.7/")
set(ARDUINO_CMAKE_PLATFORM_NAME "espressif" CACHE STRING "")
set(ARDUINO_CMAKE_PLATFORM_ARCHITECTURE "esp32" CACHE STRING "")
string(CONCAT platform_path "${ARDUINO_SDK_PATH}"
       /hardware/
       "${ARDUINO_CMAKE_PLATFORM_NAME}/"
       "${ARDUINO_CMAKE_PLATFORM_ARCHITECTURE}")
set(ARDUINO_CMAKE_PLATFORM_PATH "${platform_path}" CACHE PATH "")

# Include Arduino-CMake Toolchain
set(CMAKE_TOOLCHAIN_FILE contrib/CMake-Framework/Arduino-Toolchain.cmake)
```
But then we run into the issue that all the tools are incorrect as seen in the `_find_required_programs`:
```
function(_find_required_programs)

    # Find ASM compiler
```

**So the question is:**
- Should these not be loaded from `platform.txt` which seems to reference the _correct_ places for such things in `compiler.c.cmd`, `compiler.cpp.cmd` etc?
- Is this expected behaviour?
- Is adding boards with non avr architecture supported/planned/PR welcome? :) 

Background info in my case the `platform.txt`'s contain:
In `espressif/esp32/platform.txt`:
```
compiler.c.cmd=xtensa-esp32-elf-gcc
compiler.c.flags=-std=gnu99 -Os -g3 -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wpointer-arith {compiler.warning_flags} -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -Wno-old-style-declaration -MMD -c

compiler.cpp.cmd=xtensa-esp32-elf-g++
compiler.cpp.flags=-std=gnu++11 -fno-exceptions -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib {compiler.warning_flags} -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti -MMD -c
....continued
```

In the `arduino/avrplatform.txt`:
```
compiler.c.cmd=avr-gcc
compiler.c.flags=-c -g -Os {compiler.warning_flags} -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects
compiler.c.elf.flags={compiler.warning_flags} -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections
compiler.c.elf.cmd=avr-gcc
compiler.S.flags=-c -g -x assembler-with-cpp -flto -MMD
compiler.cpp.cmd=avr-g++
compiler.cpp.flags=-c -g -Os {compiler.warning_flags} -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno
..... continued
```

toshi38, 2018-10-24:
It looks like https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification has a pretty good description of which files define what.

It seems like a good start would be to search here for the avr things instead of requiring them to be installed globally and then expand that to search all architectures?

MrPointer, 2018-10-26:
Hello @toshi38 and thank you for all you efforts to dig into this and find a solution!
Unfortunately though, 3rd party platforms such as ESP are currently **not supported**. If you were mistaken by the Readme, I've updated it now to make this clearer.

When the time comes and I'll have the opportunity to add this feature, I'll definitely draw insight from your ideas here since they're all correct 😃

toshi38, 2018-10-28:
Thanks for the update!

tschaffter, 2020-02-15:
Hi, is there any update regarding the support of `3rd Party platforms (such as ESP32, pinoccio, etc.)`?

Thanks for your contribution to this nice project!

MrPointer, 2020-02-16:
I'm afraid not.
Moreover, I've abandoned this project a while ago, and I'm looking for new active maintainers.