arduino-cmake/Arduino-CMake-NG #53: Toolchain can't find avrdude

tobiashienzsch, 2018-10-26:
Hi,

**OS:** Ubuntu 18.04LTS & Arch Linux (kernel 4.19)

I'm having the same issue on Arch & Ubuntu
I installed all the dependencies listed in the wiki. I made sure that avrdude is installed with:
```sh
# which avrdude
/usr/bin/avrdude
```

then I went into the examples folder and created a build directory:
```sh
cd Arduino-CMake-NG/examples
mkdir build && cd build
```

The problem is that when running:
```sh
cmake -DCMAKE_TOOLCHAIN_FILE=../../cmake/Arduino-Toolchain.cmake ..
```
I get the following error:

```
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/share/arduino/hardware/tools/avr/bin/avr-gcc
CMake Error at /home/tobante/Developer/LightControl/Arduino-CMake-NG/cmake/Platform/System/AvrToolsFinder.cmake:43 (message):
  avrdude program is required by the toolchain but can't be found
Call Stack (most recent call first):
  /home/tobante/Developer/LightControl/Arduino-CMake-NG/cmake/Platform/System/BuildSystemInitializer.cmake:11 (find_tool_avrdude_configuration)
  /home/tobante/Developer/LightControl/Arduino-CMake-NG/cmake/Platform/System/BuildSystemInitializer.cmake:22 (find_required_platform_tools)
  /home/tobante/Developer/LightControl/Arduino-CMake-NG/cmake/Platform/Arduino.cmake:34 (initialize_build_system)
  /usr/share/cmake-3.10/Modules/CMakeSystemSpecificInformation.cmake:26 (include)
  CMakeLists.txt:3 (project)

-- Configuring incomplete, errors occurred!
See also "/home/tobante/Developer/LightControl/Arduino-CMake-NG/examples/build/CMakeFiles/CMakeOutput.log".
```

MrPointer, 2018-10-26:
Hi @tobiashienzsch and welcome to the project!
Where did you install your Arduino SDK? Is it one of the suggested paths? Because if not, you should set an env variable pointing to the absolute path of the SDK.

Besides, you should probably know that **archlinux** has a special option to support it (I'm afraid I didn't mention it anywhere in the docs 😞 ) - You can try to turn on the `USE_ARCHLINUX_BUILTIN_SUPPORT` option, preferably on the command-line. I can't tell whether it'll solve your current issue, that's what my question is for, but in general it's a good idea to use this option when using **archlinux**.

tobiashienzsch, 2018-10-26:
Hi @MrPointer 
thanks for the quick reply. 

All dependencies except for the SDK where installed with the package manager on both platforms.

On ArchLinux I installed the SDK using the pkg manager. And it looks like CMake found the path since in the output it shows:

```sh
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/share/arduino/hardware/tools/avr/bin/avr-gcc
```
On Ubuntu I installed via the install.sh that comes with the Linux 64bit download from arduino.cc

MrPointer, 2018-10-27:
It seems that the framework doesn't search the **avrdude** tool anywhere else outside the SDK. Given that you've installed it using a package manager, it's probably located somewhere else.

I don't have much time to fix this bug now, hopefully will resolve it in the upcoming days.
Thanks for reporting!

tobiashienzsch, 2018-11-06:
Here is the output of your command:

```sh

/usr/share/arduino/hardware/archlinux-arduino/avr/variants/circuitplay32u4:
total 12
-rw-r--r-- 1 root root 9845 Oct  9 15:52 pins_arduino.h

/usr/share/arduino/hardware/archlinux-arduino/avr/variants/eightanaloginputs:
total 4
-rw-r--r-- 1 root root 969 Oct  9 15:52 pins_arduino.h

/usr/share/arduino/hardware/archlinux-arduino/avr/variants/ethernet:
total 8
-rw-r--r-- 1 root root 6446 Oct  9 15:52 pins_arduino.h

/usr/share/arduino/hardware/archlinux-arduino/avr/variants/gemma:
total 4
-rw-r--r-- 1 root root 3244 Oct  9 15:52 pins_arduino.h

/usr/share/arduino/hardware/archlinux-arduino/avr/variants/leonardo:
total 12
-rw-r--r-- 1 root root 9870 Oct  9 15:52 pins_arduino.h

/usr/share/arduino/hardware/archlinux-arduino/avr/variants/mega:
total 16
-rw-r--r-- 1 root root 13685 Oct  9 15:52 pins_arduino.h

/usr/share/arduino/hardware/archlinux-arduino/avr/variants/micro:
total 4
-rw-r--r-- 1 root root 1190 Oct  9 15:52 pins_arduino.h

/usr/share/arduino/hardware/archlinux-arduino/avr/variants/robot_control:
total 12
-rw-r--r-- 1 root root 9232 Oct  9 15:52 pins_arduino.h

/usr/share/arduino/hardware/archlinux-arduino/avr/variants/robot_motor:
total 12
-rw-r--r-- 1 root root 8985 Oct  9 15:52 pins_arduino.h

/usr/share/arduino/hardware/archlinux-arduino/avr/variants/standard:
total 8
-rw-r--r-- 1 root root 6437 Oct  9 15:52 pins_arduino.h

/usr/share/arduino/hardware/archlinux-arduino/avr/variants/yun:
total 4
-rw-r--r-- 1 root root 1856 Oct  9 15:52 pins_arduino.h
```

I tried again after a couple of days on a new arch install. This time I had to set the ASM compiler for cmake manually with ```sh export ASM=/usr/bin/avr-gcc ``` before I got to the error from before.

The error throws in the ```AvrToolsFinder.cmake``` file on line 43 it looks for the avrdude.conf in ```etc```. Which on my system is present.

tobiashienzsch, 2018-11-06:
@MrPointer On what system are you developing? I don't really mind changing distros as I'm trying it out in a VM anyway. As soon as I get it running once, future debugging will be a lot easier.

MrPointer, 2018-11-07:
@tobiashienzsch Personally I develop on Windows, but our official tests are done against Ubuntu.
I bet it'll work as expected on Ubuntu, however, I still want to try and fix that issue on archlinux to let other users who don't have the ability to simply switch VMs to enjoy the framework as well.

As to my request, I'm sorry, but you've attached the listing of the wrong path!
I asked for `/usr/share/arduino/hardware`, whereas you've run the command on `/usr/share/arduino/hardware/archlinux-arduino/avr/variants`. 
Maybe the output was too large to embed it in a single block inside a comment? Try to create **gist** next time and provide a link to it in the comment, it will definitely work.
Thanks again!

tobiashienzsch, 2018-11-07:
@MrPointer here is the correct output: [gist](https://gist.github.com/tobiashienzsch/3469904dc140fd610f40b83df6a7da29)

tobiashienzsch, 2018-11-08:
I played around for a little while but couldn't solve it. I then started from scratch again, this time downloading the arduino-sdk directly from the website. After extracting it in my home folder I exported the path with ``` export ARDUINO_SDK_PATH="$HOME/Developer/lib/arduino-1.8.7" ```.

This time the examples build with no problems. The ``` USE_ARCHLINUX_BUILTIN_SUPPORT ``` flag was not used.

tobiashienzsch, 2018-11-08:
One quick follow up question: How do I add new platforms to the SDK?
I have a mkr1000 board. It needs the [ArduinoCore-samd](https://github.com/arduino/ArduinoCore-samd) repository. I tried cloning it to ``` $ARDUINO_SDK_PATH/hardware/arduino ``` next to ```avr```.

And then tried building, with  the following lines in ``` CMakeLists.txt ```:

```get_board_id(board_id MKR1000)``` and ``` get_board_id(board_id MKR1000 cortex-m0plus) ```

Both did not work. I'm probably missing something, but I couldn't find anything in the wiki.

MrPointer, 2018-11-09:
@tobiashienzsch I'll relate to your latest comment first - The truth is you currently can't use 3rd-party platforms since these are not supported yet!
However, support for these is one of the project's goals, so whenever it reaches v1.0, it'll definitely have support for that as part of it 😃 

As to your 1st comment, which seems to resolve the issue - It's good to know that your issue has been solved, however, it's still intriguing why and what exactly did solve it...
Nevertheless, I looked at your attached gist and it seems as there's no `tools` sub-directory under the `hardware` directory which I've asked you to list. This is probably how archlinux works, and it implies that the entire build system (**gcc, avrdude**, etc.) should be manually provided, but currently this isn't supported by the framework as it assumes these exist under the `hardware/tools` sub-directory.
In fact, supporting this is part of supporting 3rd party platforms, and maybe even the most fundamental step in it.

I'm planning on supporting it as I already said here, and I hope you could help and test the feature when it's ready!

closed 2018-11-09 by MrPointer

Dominaezzz, 2019-11-18:
This has been fixed in PR but when will it be released? (I also tried master branch but ran into different issues).

MrPointer, 2019-11-19:
@Dominaezzz Hi, sorry to disappoint but as of now probably never...
Currently, I'm the only maintainer, yet I've abandoned this project probably for good.
This PR has indeed been merged to **master**, but as this branch is also the active development branch, so it's not in a very stable state right now.

cebas, 2019-11-19:
I was using this project with great success! Thanks @MrPointer !
But now that it's going to be abandoned, I started to look for alternatives. I've found PlatformIO and it's working fine with my Arduino Uno projects. I recommend everyone to give it a try!  https://platformio.org/
Cheers

MrPointer, 2019-11-19:
@cebas PlatformIO has always been a replacement, if not a better alternative, to this project.
One of the main reasons I haven't used it myself was their lack of support for offline development, which is crucial to my current organization. However, it seems to be supported now in the new version 4.1.0 :raised_hands:

PR #81 (2019-04-22): Arch linux support
Adds support for the Arduino IDE available from the [arduino](https://www.archlinux.org/packages/community/x86_64/arduino/) and [arduino-avr-core](https://www.archlinux.org/packages/community/any/arduino-avr-core/) arch linux repositories.  All examples now compile assuming you've installed the required Arduino libraries and set `AUTO_SET_SKETCHBOOK_PATH=ON`.

Hopefully fixes #53 (for Arch Linux, Ubuntu ships an ancient version 1.0.5 of the IDE, untested on Fedora)

AFAIK Fedora does something similar to Arch but I haven't tested that yet so if someone with Fedora could test [... cut]