arduino-cmake/Arduino-CMake-NG #54: $<COMPILE_LANGUAGE:...> Unknown language - RASPBIAN

doleron, 2018-11-08:
Hi,
my OS is Raspbian stretch
0 - I followed the steps into the "installation":
```
sudo apt-get install gcc-avr
sudo apt-get install binutils-avr
sudo apt-get install avr-libc 
sudo apt-get install avrdude
```
and installed arduino 1.8.7 from the arm tarball.
I installed cmake 3.8.2 from sources
1 - donwloaded Arduino-CMake-NG and extracted it in /home/pi
2 - `cd ~/Arduino-CMake-NG-master/examples/blink-example`
3 - `cmake -DCMAKE_TOOLCHAIN_FILE=/home/pi/Arduino-CMake-NG-master/cmake/Arduino-Toolchain.cmake .`
This is the output:
```
pi@raspberrypi:~/Arduino-CMake-NG-master/examples/blink-example $ cmake -DCMAKE_TOOLCHAIN_FILE=/home/pi/Arduino-CMake-NG-master/cmake/Arduino-Toolchain.cmake .
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Arduino SDK version 1.8.7: /usr/share/arduino
-- Determined Platform Header: /usr/share/arduino/hardware/arduino/avr/cores/arduino/Arduino.h
-- Arduino SDK version 1.8.7: /usr/share/arduino
-- Determined Platform Header: /usr/share/arduino/hardware/arduino/avr/cores/arduino/Arduino.h
-- Check for working C compiler: /usr/share/arduino/hardware/tools/avr/bin/avr-gcc
-- Check for working C compiler: /usr/share/arduino/hardware/tools/avr/bin/avr-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Arduino SDK version 1.8.7: /usr/share/arduino
-- Determined Platform Header: /usr/share/arduino/hardware/arduino/avr/cores/arduino/Arduino.h
-- Check for working CXX compiler: /usr/share/arduino/hardware/tools/avr/bin/avr-g++
-- Check for working CXX compiler: /usr/share/arduino/hardware/tools/avr/bin/avr-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
CMake Error at /home/pi/Arduino-CMake-NG-master/cmake/Platform/Other/TargetFlagsManager.cmake:13 (target_compile_options):
  Error evaluating generator expression:

    $<COMPILE_LANGUAGE:ASM>

  $<COMPILE_LANGUAGE:...> Unknown language.
Call Stack (most recent call first):
  /home/pi/Arduino-CMake-NG-master/cmake/Platform/Other/TargetFlagsManager.cmake:35 (_set_target_language_flags)
  /home/pi/Arduino-CMake-NG-master/cmake/Platform/Targets/CoreLibTarget.cmake:67 (set_compiler_target_flags)
  /home/pi/Arduino-CMake-NG-master/cmake/Platform/Targets/CoreLibTarget.cmake:112 (_set_core_lib_flags)
  /home/pi/Arduino-CMake-NG-master/cmake/Platform/Targets/ExecutableTarget.cmake:15 (add_arduino_core_lib)
  /home/pi/Arduino-CMake-NG-master/cmake/Platform/Targets/ArduinoExampleTarget.cmake:17 (add_arduino_executable)
  CMakeLists.txt:7 (add_arduino_example)

CMake Error at /home/pi/Arduino-CMake-NG-master/cmake/Platform/Other/TargetFlagsManager.cmake:13 (target_compile_options):
  Error evaluating generator expression:

    $<COMPILE_LANGUAGE:ASM>

  $<COMPILE_LANGUAGE:...> Unknown language.
Call Stack (most recent call first):
  /home/pi/Arduino-CMake-NG-master/cmake/Platform/Other/TargetFlagsManager.cmake:35 (_set_target_language_flags)
  /home/pi/Arduino-CMake-NG-master/cmake/Platform/Targets/CoreLibTarget.cmake:67 (set_compiler_target_flags)
  /home/pi/Arduino-CMake-NG-master/cmake/Platform/Targets/CoreLibTarget.cmake:112 (_set_core_lib_flags)
  /home/pi/Arduino-CMake-NG-master/cmake/Platform/Targets/ExecutableTarget.cmake:15 (add_arduino_core_lib)
  /home/pi/Arduino-CMake-NG-master/cmake/Platform/Targets/ArduinoExampleTarget.cmake:17 (add_arduino_executable)
  CMakeLists.txt:7 (add_arduino_example)

CMake Error at /home/pi/Arduino-CMake-NG-master/cmake/Platform/Other/TargetFlagsManager.cmake:13 (target_compile_options):
  Error evaluating generator expression:

    $<COMPILE_LANGUAGE:ASM>

  $<COMPILE_LANGUAGE:...> Unknown language.
Call Stack (most recent call first):
  /home/pi/Arduino-CMake-NG-master/cmake/Platform/Other/TargetFlagsManager.cmake:35 (_set_target_language_flags)
  /home/pi/Arduino-CMake-NG-master/cmake/Platform/Other/TargetFlagsManager.cmake:70 (set_compiler_target_flags)
  /home/pi/Arduino-CMake-NG-master/cmake/Platform/Targets/ExecutableTarget.cmake:17 (set_executable_target_flags)
  /home/pi/Arduino-CMake-NG-master/cmake/Platform/Targets/ArduinoExampleTarget.cmake:17 (add_arduino_executable)
  CMakeLists.txt:7 (add_arduino_example)

CMake Error at /home/pi/Arduino-CMake-NG-master/cmake/Platform/Targets/CoreLibTarget.cmake:116 (target_link_libraries):
  Error evaluating generator expression:

    $<COMPILE_LANGUAGE:ASM>

  $<COMPILE_LANGUAGE:...> Unknown language.
Call Stack (most recent call first):
  /home/pi/Arduino-CMake-NG-master/cmake/Platform/Targets/ExecutableTarget.cmake:15 (add_arduino_core_lib)
  /home/pi/Arduino-CMake-NG-master/cmake/Platform/Targets/ArduinoExampleTarget.cmake:17 (add_arduino_executable)
  CMakeLists.txt:7 (add_arduino_example)

-- Generating done
-- Build files have been written to: /home/pi/Arduino-CMake-NG-master/examples/blink-example

```
Any ideas about what I did wrong?

tobiashienzsch, 2018-11-08:
Hi,

I installed almost exactly the same setup today, CMake which I build from source is newer, version ``` 3.13 ```. I also downloaded the "arduino-sdk" from the website & extracted it to my home folder I additionally set the ```ARDUINO_SDK_PATH ``` env variable, maybe this helps. Try this command: ``` export ARDUINO_SDK_PATH= "path/to/arduino-sdk" ``` or set it in your ```.bashrc ```.

doleron, 2018-11-09:
Hi, @tobiashienzsch .
I just upgraded cmake for 3.13.0-rc3 and then the examples worked. Thank you. Sidenote: Since than arduino sdk was installed in /usr/share, the command `export ARDUINO_SDK_PATH= "path/to/arduino-sdk"` didn't seem to make any difference in the end result.

closed 2018-11-09 by doleron