arduino-cmake/Arduino-CMake-NG #61: Custom Library Folder

Alexivia, 2018-12-05:
Hi,

I'm trying to use this tool-chain to build my complex Arduino project with multiple sources and custom libraries. I have a folder apart from the SDK's sketchbook where I store all the custom libraries I created along the past years, and are used across different projects... However, when I try to compile my project with this framework, it doesn't automatically find these libraries in this folder, though it is sym-linked to a "*libraries*" folder on the CMake project root... If I manually link them using "*find_arduino_library*" and "*link_arduino_library*" it finds them and compiles them, however I have numerous libraries and this manually linking becomes cumbersome... is there an automatic way of doing this?

Apart from this, very nice work developing and maintaining this project! Thank you for your time!

Regards,
Alexandre

MrPointer, 2018-12-07:
@Alexivia Hello and welcome to the project, I appreciate your words very much!
Regarding your concern - I have some trouble realizing exactly what you're expecting from the framework.

Libraries, as the framework treats them, are simply folders that conform to a certain standard implied by Arduino. Not all of them have to conform, especially when they're custom libraries as in your case, but **all of them** _have to be found_ - What it means is that you should always specify which libraries are to be included in your project using **CMake** code.

Looking back at your description - It looks like you're looking for a way to request the framework to treat all of the sub-directories of a given directory as libraries that are going to be used. 
And, since you have a lot of libraries, you probably won't use **all** of them every single time - What makes this request somewhat uncomfortable, from an automation perspective.
Eventually, such a feature will cause a code-size blast, as all libraries will be built and statically linked to targets, maybe even a single target, and that of course increases code size (and memory). No embedded developer ever wants that, is it?

Before we continue discussing alternatives, please let me know if this indeed was your intention.

Alexivia, 2018-12-10:
@MrPointer Thank you for your reply!

Though my libraries are custom, I comply to the Arduino standard by having a directory with the library name and the source files inside a `src` sub-directory. The framework makes this recursive search with the standard Arduino libraries, by compiling and linking only the requested libraries on the code. My suggestion was to do this selective recursive search inside a user-provided directory with Arduino-compliant libraries that would only be compiled and linked if found mentioned in `#include <.h>` statements on the provided source code.

Once again, I can make my projects correctly comply using the framework as-is, however it is a cumbersome job to make a specific set of cmake include and link statements for each different project, moreover to link them to the correct target.

Thank you again for your feedback!
Regards,
Alexandre

MrPointer, 2018-12-11:
Well, your desired feature is simply not going to happen anytime soon, sorry.
The purpose of the framework is to provide Arduino developers an easy way to maintain their projects using CMake, as it's a CMake framework, and by no means it should behave similar to what the Arduino IDE does. In other words, this is just something that can't be done well with CMake as this is not the right tool for this job, and is completely out of the scope of the framework, at least for now.
I understand the inconvenience, however, this kind of support is more suitable for a plugin.

For example, take **VS Code** - It could have an extension/plugin for **Arduino-CMake** that automatically generates matching CMake scripts that make use of the framework. Moreover, it could read your source files, search for includes, and generate explicit CMake code to use matching libraries based on its' findings.
This sort of task is typical for a plugin, but not to a CMake framework, unfortunately.
If you like this idea, you can always put some effort and develop it! 😃

Alexivia, 2018-12-11:
I'm not an expert in CMake nor Make by any means, I just know enough to use these kind of frameworks and basic scripting... Maybe that's why I though my request was easier to implement than what you are saying... But it makes sense that an IDE plugin is best fit for this task!

Thank you for your support!