arduino-cmake/Arduino-CMake-NG #64: SPI library not found

Tyrrx, 2018-12-12:
Hello,
I have a library that uses the Arduino SPI library.
After realising #include<SPI.h> doesn't work i tried adding it maually by adding 
find_arduino_library(spi_lib SPI ${board_id})
link_arduino_library(main spi_lib ${board_id})
but I just got the error:
CMake Error at cmake/Platform/Libraries/LibrariesFinder.cmake:38 (message):
  Couldn't find library named Spi

So how do i use the SPI library located at:
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI

MrPointer, 2018-12-13:
@Tyrrx Hi there, please add some info regarding your environment as stated [here](https://github.com/arduino-cmake/Arduino-CMake-NG/blob/master/CONTRIBUTING.md#issue-signature).

stuckless, 2018-12-29:
I've hit this issue as well.  I was trying to convert from arduino-cmake to this version, but, I seem to have hit 2 issues as listed here.

1. It doesn't appear to find libraries that are listed in `ArduinoSDK/hardware/arduino/avr/libraries`
2. It converts `SPI` to `Spi`

```makefile
/home/seans/.local/share/JetBrains/Toolbox/apps/CLion/ch-0/183.4886.39/bin/cmake/linux/bin/cmake -DCMAKE_BUILD_TYPE=Debug -G "CodeBlocks - Unix Makefiles" /home/seans/git/trillium/trillium-holt-arduino
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Arduino SDK version 1.8.5: /home/seans/apps/ArduinoSDK
-- Determined Platform Header: /home/seans/apps/ArduinoSDK/hardware/arduino/avr/cores/arduino/Arduino.h
-- Arduino SDK version 1.8.5: /home/seans/apps/ArduinoSDK
-- Determined Platform Header: /home/seans/apps/ArduinoSDK/hardware/arduino/avr/cores/arduino/Arduino.h
-- Check for working C compiler: /home/seans/apps/ArduinoSDK/hardware/tools/avr/bin/avr-gcc
-- Check for working C compiler: /home/seans/apps/ArduinoSDK/hardware/tools/avr/bin/avr-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Arduino SDK version 1.8.5: /home/seans/apps/ArduinoSDK
-- Determined Platform Header: /home/seans/apps/ArduinoSDK/hardware/arduino/avr/cores/arduino/Arduino.h
-- Check for working CXX compiler: /home/seans/apps/ArduinoSDK/hardware/tools/avr/bin/avr-g++
-- Check for working CXX compiler: /home/seans/apps/ArduinoSDK/hardware/tools/avr/bin/avr-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at cmake/Platform/Libraries/LibrariesFinder.cmake:38 (message):
  Couldn't find library named Spi
Call Stack (most recent call first):
  CMakeLists.txt:50 (find_arduino_library)

CMake Error at cmake/Platform/Targets/ArduinoLibraryTarget.cmake:70 (message):
  Target doesn't exist - It must be created first!
Call Stack (most recent call first):
  CMakeLists.txt:51 (link_arduino_library)

-- Configuring incomplete, errors occurred!
See also "/home/seans/git/trillium/trillium-holt-arduino/cmake-build-debug/CMakeFiles/CMakeOutput.log".

[Finished]
```
if I hack `LibrariesFinder.cpp` and change the `find_file` `PATH` and add my `ArduinoSDK/hardware/arduino/avr/libraries` and then comment out the code that converts to pascal case... it finds the library... but then fails on issue #56.

My `LibrariesFinder.cpp` hacks are basically this..
```cmake
#    if (NOT parsed_args_3RD_PARTY)
#        convert_string_to_pascal_case(${_library_name} _library_name)
#    endif ()

    find_file(library_path
            NAMES ${_library_name}
            PATHS ${ARDUINO_SDK_LIBRARIES_PATH} ${ARDUINO_CMAKE_SKETCHBOOK_PATH} /home/seans/apps/ArduinoSDK/hardware/arduino/avr/libraries
            ${CMAKE_CURRENT_SOURCE_DIR} ${PROJECT_SOURCE_DIR}
            PATH_SUFFIXES libraries dependencies
            NO_DEFAULT_PATH
            NO_CMAKE_FIND_ROOT_PATH)
```

```makefile
/home/seans/.local/share/JetBrains/Toolbox/apps/CLion/ch-0/183.4886.39/bin/cmake/linux/bin/cmake -DCMAKE_BUILD_TYPE=Debug -G "CodeBlocks - Unix Makefiles" /home/seans/git/trillium/trillium-holt-arduino
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Arduino SDK version 1.8.5: /home/seans/apps/ArduinoSDK
-- Determined Platform Header: /home/seans/apps/ArduinoSDK/hardware/arduino/avr/cores/arduino/Arduino.h
-- Arduino SDK version 1.8.5: /home/seans/apps/ArduinoSDK
-- Determined Platform Header: /home/seans/apps/ArduinoSDK/hardware/arduino/avr/cores/arduino/Arduino.h
-- Check for working C compiler: /home/seans/apps/ArduinoSDK/hardware/tools/avr/bin/avr-gcc
-- Check for working C compiler: /home/seans/apps/ArduinoSDK/hardware/tools/avr/bin/avr-gcc -- works
-- Detecting C compiler ABI info
[... 41 lines cut ...]

file(GLOB ARINC_HDRS arinc429/holt/*.h)
file(GLOB ARINC_SRCS arinc429/holt/*.cpp)

file(GLOB DIO_HDRS dio/holt/*.h)
file(GLOB DIO_SRCS dio/holt/*.cpp)

file(GLOB RUNTIME_HDRS runtime/arduino/*.h)
file(GLOB RUNTIME_SRCS runtime/arduino/*.cpp)

file(GLOB PROG_HDRS programs/*.h)
file(GLOB PROG_SRCS programs/*.cpp)

file(GLOB MAIN_HDRS *.h)
file(GLOB MAIN_SRCS *.cpp)

set(${CMAKE_PROJECT_NAME}_SRCS  ${API_SRCS} ${ARINC_SRCS} ${DIO_SRCS} ${RUNTIME_SRCS} ${PROG_SRCS} ${MAIN_SRCS})
set(${CMAKE_PROJECT_NAME}_HDRS  ${API_HDRS} ${ARINC_HDRS} ${DIO_HDRS} ${RUNTIME_HDRS} ${PROG_HDRS} ${MAIN_HDRS})

add_arduino_executable(${CMAKE_PROJECT_NAME} ${board_id} ${${CMAKE_PROJECT_NAME}_SRCS})

find_arduino_library(spi_lib SPI ${board_id})
link_arduino_library(Arduino_Library spi_lib ${board_id})

```

MrPointer, 2018-12-29:
@stuckless Your problem is wrong usage of API, unfortunately...
Take a minute to read [this](https://github.com/arduino-cmake/Arduino-CMake-NG/wiki/Platform-Libraries) - I've added this just a few days ago (although the API has been consistent all this time)
Besides, are you using the [latest version](https://github.com/arduino-cmake/Arduino-CMake-NG/releases/tag/v0.6.1) of this framework? i.e. The top-listed in the [releases](https://github.com/arduino-cmake/Arduino-CMake-NG/releases) page.

stuckless, 2018-12-29:
@MrPointer - thx for the quick response.

So, in changing my CMakeLists.txt to be the following...
```cmake
add_arduino_executable(${CMAKE_PROJECT_NAME} ${board_id} ${${CMAKE_PROJECT_NAME}_SRCS})
link_arduino_library(${CMAKE_PROJECT_NAME} SPI ${board_id})
```
I still get the following issue :(

```cmake
CMake Error at cmake/Platform/Targets/ArduinoLibraryTarget.cmake:72 (message):
  Library target doesn't exist - It must be created first!
Call Stack (most recent call first):
  CMakeLists.txt:48 (link_arduino_library)

```
Clearly, I'm still missing something.

MrPointer, 2018-12-29:
@stuckless Actually, your'e not missing anything at all. I've just checked this on my end and found out it's clearly a bug, and moreover, a serious lack of API, which relates to #66 .

About the clone source - I'd rather not depend on the **master** branch as this is the main development branch in this repo, where release-unready code might exist, as opposed to a tagged release.

stuckless, 2018-12-30:
@MrPointer thx, I switched to the released version.  And thanks for you work this.

MrPointer, 2018-12-30:
@stuckless I just remembered that there's a dedicated function for linking platform libraries - It's called `link_platform_library` and its' API is the same as for `link_arduino_library`.
Nonetheless, it doesn't mean that this API isn't flawed and shouldn't be fixed.

stuckless, 2018-12-31:
@MrPointer thx.   I can confirm that does does work.

My CMakeLists.txt now includes the following directives, and they build and upload in a similar way as the older Arduino-CMake project.

```cmake
add_arduino_executable(${PROJECT_NAME} "${board_id}" ${${PROJECT_NAME}_sources} ${${PROJECT_NAME}_headers})
link_platform_library(${PROJECT_NAME} SPI "${board_id}")
upload_arduino_target(${PROJECT_NAME} "${board_id}" ${ARDUNIO_SERIAL_PORT})
```
Thanks again for you help with this.  I do think that it would be beneficial to make the library handling more automatic (as I'm guessing you are planning with #66 ).  I personally had no idea that SPI was a platform library vs an arduino library.  The older `Arduino-CMake` figured this out, somehow, since I didn't add anything to my project's CMakeLists.txt to directly reference it.  I do appreciate the work you've done on this.

MrPointer, 2018-12-31:
@stuckless Great to hear that, thanks.
About the API - I've known this project for a long time now, having worked on it's most popular and attractive fork prior to this "NG" project, so I know exactly how messy stuff was back in the days. Some parts were good, especially regarding usability, bad had been awful design and implementation-wise.
This project tries to address many of those issues, but I don't always get it right of course - That's what community is for 😃

closed 2018-12-31 by MrPointer