arduino-cmake/Arduino-CMake-NG #65: Problem with ARDUINO_CMAKE_PLATFORM_PROPERTIES_FILE_PATH-NOTFOUND when building Hello_World

refcodes, 2018-12-21:
I am a newby and try to get the HelloWorld-example up and running. I run Ubuntu (Windows Subsystem for Linux) and I cloned the `CMake-NG` repo and created a `CMakeLists.txt` file as follows:

```
cmake_minimum_required(VERSION 3.8.2)
project(Hello_World)
get_board_id(board_id nano atmega328)
add_arduino_executable(Hello_World ${board_id} helloWorld.cpp)
```

When I call  `cmake -DCMAKE_TOOLCHAIN_FILE=../cmake/Arduino-CMake-NG/cmake/Arduino-Toolchain.cmake ..` I get the following:

```
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Arduino SDK version 2:1.0.5+dfsg2-4
: /usr/share/arduino
-- Determined Platform Header:
CMake Error at cmake/Arduino-CMake-NG/cmake/Platform/Properties/PropertiesReader.cmake:25 (file):
  file STRINGS file
  "/home/userx/Workspaces/arduino/playground/ARDUINO_CMAKE_PLATFORM_PROPERTIES_FILE_PATH-NOTFOUND"
  cannot be read.
Call Stack (most recent call first):
  cmake/Arduino-CMake-NG/cmake/Platform/System/PlatformInitializer.cmake:26 (read_properties_from_file)
  cmake/Arduino-CMake-NG/cmake/Platform/System/PlatformInitializer.cmake:60 (initialize_platform_properties)
  cmake/Arduino-CMake-NG/cmake/Platform/System/BuildSystemInitializer.cmake:31 (initialize_arduino_platform)
  cmake/Arduino-CMake-NG/cmake/Platform/Arduino.cmake:34 (initialize_build_system)
  /usr/share/cmake-3.12/Modules/CMakeSystemSpecificInformation.cmake:26 (include)
  CMakeLists.txt:3 (project)

-- Arduino SDK version 2:1.0.5+dfsg2-4
: /usr/share/arduino
-- Determined Platform Header:
CMake Error at cmake/Arduino-CMake-NG/cmake/Platform/Properties/PropertiesReader.cmake:25 (file):
  file STRINGS file
  "/home/userx/Workspaces/arduino/playground/ARDUINO_CMAKE_PLATFORM_PROPERTIES_FILE_PATH-NOTFOUND"
  cannot be read.
Call Stack (most recent call first):
  cmake/Arduino-CMake-NG/cmake/Platform/System/PlatformInitializer.cmake:26 (read_properties_from_file)
  cmake/Arduino-CMake-NG/cmake/Platform/System/PlatformInitializer.cmake:60 (initialize_platform_properties)
  cmake/Arduino-CMake-NG/cmake/Platform/System/BuildSystemInitializer.cmake:31 (initialize_arduino_platform)
  cmake/Arduino-CMake-NG/cmake/Platform/Arduino.cmake:34 (initialize_build_system)
  /usr/share/cmake-3.12/Modules/CMakeCInformation.cmake:66 (include)
  CMakeLists.txt:3 (project)

-- Check for working C compiler: /usr/share/arduino/hardware/tools/avr/bin/avr-gcc
CMake Error at /home/userx/Workspaces/arduino/playground/cmake/Arduino-CMake-NG/cmake/Platform/Properties/PropertiesReader.cmake:25 (file):
  file STRINGS file
  "/home/userx/Workspaces/arduino/playground/bin/CMakeFiles/CMakeTmp/ARDUINO_CMAKE_PLATFORM_PROPERTIES_FILE_PATH-NOTFOUND"
  cannot be read.
Call Stack (most recent call first):
  /home/userx/Workspaces/arduino/playground/cmake/Arduino-CMake-NG/cmake/Platform/System/PlatformInitializer.cmake:26 (read_properties_from_file)
  /home/userx/Workspaces/arduino/playground/cmake/Arduino-CMake-NG/cmake/Platform/System/PlatformInitializer.cmake:60 (initialize_platform_properties)
  /home/userx/Workspaces/arduino/playground/cmake/Arduino-CMake-NG/cmake/Platform/System/BuildSystemInitializer.cmake:31 (initialize_arduino_platform)
  /home/userx/Workspaces/arduino/playground/cmake/Arduino-CMake-NG/cmake/Platform/Arduino.cmake:34 (initialize_build_system)
  /usr/share/cmake-3.12/Modules/CMakeSystemSpecificInformation.cmake:26 (include)
  /home/userx/Workspaces/arduino/playground/bin/CMakeFiles/CMakeTmp/CMakeLists.txt:3 (project)

CMake Error at /home/userx/Workspaces/arduino/playground/cmake/Arduino-CMake-NG/cmake/Platform/Properties/PropertiesReader.cmake:25 (file):
  file STRINGS file
  "/home/userx/Workspaces/arduino/playground/bin/CMakeFiles/CMakeTmp/ARDUINO_CMAKE_PLATFORM_PROPERTIES_FILE_PATH-NOTFOUND"
  cannot be read.
Call Stack (most recent call first):
  /home/userx/Workspaces/arduino/playground/cmake/Arduino-CMake-NG/cmake/Platform/System/PlatformInitializer.cmake:26 (read_properties_from_file)
  /home/userx/Workspaces/arduino/playground/cmake/Arduino-CMake-NG/cmake/Platform/System/PlatformInitializer.cmake:60 (initialize_platform_properties)
  /home/userx/Workspaces/arduino/playground/cmake/Arduino-CMake-NG/cmake/Platform/System/BuildSystemInitializer.cmake:31 (initialize_arduino_platform)
  /home/userx/Workspaces/arduino/playground/cmake/Arduino-CMake-NG/cmake/Platform/Arduino.cmake:34 (initialize_build_system)
  /usr/share/cmake-3.12/Modules/CMakeCInformation.cmake:66 (include)
  /home/userx/Workspaces/arduino/playground/bin/CMakeFiles/CMakeTmp/CMakeLists.txt:3 (project)

CMake Warning:
  Value of ARDUINO_CMAKE_SDK_VERSION contained a newline; truncating

CMake Warning:
  Value of ARDUINO_CMAKE_SDK_VERSION_PATCH contained a newline; truncating

CMake Warning:
  Value of runtime_ide_version contained a newline; truncating

CMake Error at /usr/share/cmake-3.12/Modules/CMakeTestCCompiler.cmake:37 (try_compile):
  Failed to configure test project build system.
Call Stack (most recent call first):
  CMakeLists.txt:3 (project)

-- Configuring incomplete, errors occurred!
See also "/home/userx/Workspaces/arduino/playground/bin/CMakeFiles/CMakeOutput.log".
CMake Warning:
  Value of ARDUINO_CMAKE_SDK_VERSION contained a newline; truncating

CMake Warning:
  Value of ARDUINO_CMAKE_SDK_VERSION_PATCH contained a newline; truncating

CMake Warning:
  Value of runtime_ide_version contained a newline; truncating
```

My cmake version is `3.12.1`.

Any help is appreciated and many thanks beforehand.

JonasProgrammer, 2018-12-21:
@MrPointer To still be at least somewhat useful to the project, the version you're seeing is likely a debian package version. From the version string it should be the Debian Jessie branch.

Basically it is an Arduino SDK 1.0.5 with some patches to match Debian's interpretation of the FHS. You can also get the binaries here to compare them: https://packages.debian.org/jessie/arduino

MrPointer, 2018-12-22:
@JonasProgrammer Oh good to see you're alive and well buddy! Thanks for the insight 😄 

@refcodes As for you pal - It seems as if your Arduino version is very old and incompatible with this project. It's also listed in the [requirements](https://github.com/arduino-cmake/Arduino-CMake-NG#requirements), so please update if you'd like to use this framework.