arduino-cmake/Arduino-CMake-NG #71: Include library header from another library

ustisha, 2019-01-22:
Hello, I can't understand where problem.
I prepared example project  https://github.com/ustisha/CMake

OS: Windows 10
Platform: Arduino
Platform SDK Version: [Arduino SDK version 1.8.8]
IDE: CLion
MinGW: 5.0
CMake: Bundled 3.13.2

Libararies is 
* https://github.com/milesburton/Arduino-Temperature-Control-Library
* https://github.com/PaulStoffregen/OneWire

Main problem is 
```[ 90%] Building CXX object test1/CMakeFiles/DallasTemperature_Lib.dir/libraries/DallasTemperature/DallasTemperature.cpp.obj
In file included from C:\Arduino\CMake\test1\libraries\DallasTemperature\DallasTemperature.cpp:6:0:
C:\Arduino\CMake\test1\libraries\DallasTemperature\DallasTemperature.h:25:21: fatal error: OneWire.h: No such file or directory
compilation terminated.
```

When i change include in _test1/libraries/DallasTemperature/DallasTemperature.h_ from `<OneWire.h>` to `#include <../OneWire/OneWire.h>` compile success. Should I change it? Will be great if i can use libraries wihtout changes (as is)

MrPointer, 2019-01-22:
Hello @ustisha, please attach your **CMakeLists.txt** file as well.

ustisha, 2019-01-22:
Here is two files 
[CMakeLists.txt](https://github.com/arduino-cmake/Arduino-CMake-NG/files/2784166/CMakeLists.txt)
[CMakeLists.txt](https://github.com/arduino-cmake/Arduino-CMake-NG/files/2784167/CMakeLists.txt)
Project structure you can check in https://github.com/ustisha/CMake

MrPointer, 2019-01-22:
@ustisha Actually you're right, my bad.
Usually people embed it in the comment like a code snippet, so I'm used to that pattern.

Regarding your files, I can see 2 problems:
1. You've set the toolchain file path inside the **CMakeLists** - It's not recommended and not the way to do it in this framework. Please refer to [this wiki page](https://github.com/arduino-cmake/Arduino-CMake-NG/wiki/Installation#preparing-environment) for more info.
2. It appears from your attached error message that your libraries are located in a non-standard location, that the framework has no way of finding by its own. For that, please refer to the [libraries wiki section](https://github.com/arduino-cmake/Arduino-CMake-NG/wiki/Libraries).

ustisha, 2019-01-23:
@MrPointer Thank you for quick answers.
1. You are right, removed it.
2. Maybe i missunderstood wiki, but as I see _libraries_ (or _dependencies_) directory must be lokated in `PROJECT_SOURCE_DIR` (for example in my case it is `C:/Arduino/CMake/test1`) and library directory has name as .cpp and .h files. Include `<OneWire.h>` and `<DallasTemperature.h>` in main.cpp works great. Eg
- CMakeLists.txt
- test1/CMakeLists.txt
- test1/main.cpp
- test1/libraries/DallasTemperature/DallasTemperature.h
- test1/libraries/OneWire/OneWire.h

```
message(${ARDUINO_SDK_LIBRARIES_PATH})
message(${PROJECT_SOURCE_DIR})
```
```
C:/Program Files (x86)/Arduino/libraries
C:/Arduino/CMake/test1
```

CMake output
```
C:\JetBrains\apps\CLion\ch-0\183.4886.39\bin\cmake\win\bin\cmake.exe -DCMAKE_BUILD_TYPE= -DCMAKE_TOOLCHAIN_FILE=arduino-cmake-ng/cmake/Arduino-Toolchain.cmake -G "CodeBlocks - MinGW Makefiles" C:\Arduino\CMake
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Arduino SDK version 1.8.8: C:/Program Files (x86)/Arduino
-- Determined Platform Header: C:/Program Files (x86)/Arduino/hardware/arduino/avr/cores/arduino/Arduino.h
-- Arduino SDK version 1.8.8: C:/Program Files (x86)/Arduino
-- Determined Platform Header: C:/Program Files (x86)/Arduino/hardware/arduino/avr/cores/arduino/Arduino.h
-- Check for working C compiler: C:/Program Files (x86)/Arduino/hardware/tools/avr/bin/avr-gcc.exe
-- Check for working C compiler: C:/Program Files (x86)/Arduino/hardware/tools/avr/bin/avr-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Arduino SDK version 1.8.8: C:/Program Files (x86)/Arduino
-- Determined Platform Header: C:/Program Files (x86)/Arduino/hardware/arduino/avr/cores/arduino/Arduino.h
-- Check for working CXX compiler: C:/Program Files (x86)/Arduino/hardware/tools/avr/bin/avr-g++.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Arduino/hardware/tools/avr/bin/avr-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
C:/Program Files (x86)/Arduino/libraries
C:/Arduino/CMake/test1
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Arduino/CMake/cmake-build-default-mingw

[Finished]
```

Build output
```
====================[ Build | test1 | Default-MinGW ]===========================
C:\JetBrains\apps\CLion\ch-0\183.4886.39\bin\cmake\win\bin\cmake.exe --build C:\Arduino\CMake\cmake-build-default-mingw --target test1 -- -j 2
[ 80%] Built target mini_atmega328_core_lib
[ 83%] Building CXX object test1/CMakeFiles/DallasTemperature.dir/C_/Program_Files_(x86)/Arduino/libraries/DallasTemperature/DallasTemperature.cpp.obj
[ 90%] Built target OneWire
In file included from C:\Program Files (x86)\Arduino\libraries\DallasTemperature\DallasTemperature.cpp:6:0:
C:\Program Files (x86)\Arduino\libraries\DallasTemperature\DallasTemperature.h:25:21: fatal error: OneWire.h: No such file or directory
compilation terminated.
test1\CMakeFiles\DallasTemperature.dir\build.make:61: recipe for target 'test1/CMakeFiles/DallasTemperature.dir/C_/Program_Files_(x86)/Arduino/libraries/DallasTemperature/DallasTemperature.cpp.obj' failed
mingw32-make.exe[3]: *** [test1/CMakeFiles/DallasTemperature.dir/C_/Program_Files_(x86)/Arduino/libraries/DallasTemperature/DallasTemperature.cpp.obj] Error 1
mingw32-make.exe[2]: *** [test1/CMakeFiles/DallasTemperature.dir/all] Error 2
CMakeFiles\Makefile2:208: recipe for target 'test1/CMakeFiles/DallasTemperature.dir/all' failed
CMakeFiles\Makefile2:109: recipe for target 'test1/CMakeFiles/test1.dir/rule' failed
mingw32-make.exe[1]: *** [test1/CMakeFiles/test1.dir/rule] Error 2
mingw32-make.exe: *** [test1] Error 2
Makefile:117: recipe for target 'test1' failed
```

MrPointer, 2019-03-01:
@ustisha Sorry again for the long delay, was sure I was the last one commented here 😆 
Anyways, could you somehow share your entire project? I need to take a look at your structure, thanks.

ustisha, 2019-05-14:
Hello back @MrPointer ! I'm do some research as much as I can. For now still do not understand few things.
1. I updated my example project
2. Adding libs works as I described in CMakeLists.txt I added some debug message and i see adding is succesfull 
```
C:\JetBrains\apps\CLion\ch-0\191.6707.69\bin\cmake\win\bin\cmake.exe -DCMAKE_BUILD_TYPE=Debug -DCMAKE_TOOLCHAIN_FILE=arduino-cmake-ng/cmake/Arduino-Toolchain.cmake -G "CodeBlocks - MinGW Makefiles" C:\Arduino\CMake
-- Arduino SDK version 1.8.8: C:/Program Files (x86)/Arduino
-- Determined Platform Header: C:/Program Files (x86)/Arduino/hardware/arduino/avr/cores/arduino/Arduino.h
-- Arduino SDK version 1.8.8: C:/Program Files (x86)/Arduino
-- Determined Platform Header: C:/Program Files (x86)/Arduino/hardware/arduino/avr/cores/arduino/Arduino.h
-- Arduino SDK version 1.8.8: C:/Program Files (x86)/Arduino
-- Determined Platform Header: C:/Program Files (x86)/Arduino/hardware/arduino/avr/cores/arduino/Arduino.h
add_arduino_library OneWire_Lib - mini.atmega328 - C:/Arduino/CMake/libraries/Onewire/OneWire.h;C:/Arduino/CMake/libraries/Onewire/OneWire.cpp
add_arduino_library DallasTemperature_Lib - mini.atmega328 - C:/Arduino/CMake/libraries/Dallastemperature/DallasTemperature.h;C:/Arduino/CMake/libraries/Dallastemperature/DallasTemperature.cpp
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Arduino/CMake/cmake-build-debug

[Finished]
```
But build still failed 
```
====================[ Build | test1 | Debug ]===================================
C:\JetBrains\apps\CLion\ch-0\191.6707.69\bin\cmake\win\bin\cmake.exe --build C:\Arduino\CMake\cmake-build-debug --target test1 -- -j 4
[ 80%] Built target mini_atmega328_core_lib
Scanning dependencies of target DallasTemperature_Lib
[ 90%] Building CXX object test1/CMakeFiles/DallasTemperature_Lib.dir/__/libraries/DallasTemperature/DallasTemperature.cpp.obj
[ 90%] Built target OneWire_Lib
In file included from C:\Arduino\CMake\libraries\DallasTemperature\DallasTemperature.cpp:6:0:
C:\Arduino\CMake\libraries\DallasTemperature\DallasTemperature.h:25:21: fatal error: OneWire.h: No such file or directory
compilation terminated.
mingw32-make.exe[3]: *** [test1/CMakeFiles/DallasTemperature_Lib.dir/__/libraries/DallasTemperature/DallasTemperature.cpp.obj] Error 1
test1\CMakeFiles\DallasTemperature_Lib.dir\build.make:61: recipe for target 'test1/CMakeFiles/DallasTemperature_Lib.dir/__/libraries/DallasTemperature/DallasTemperature.cpp.obj' failed
mingw32-make.exe[2]: *** [test1/CMakeFiles/DallasTemperature_Lib.dir/all] Error 2
CMakeFiles\Makefile2:95: recipe for target 'test1/CMakeFiles/DallasTemperature_Lib.dir/all' failed
mingw32-make.exe[1]: *** [test1/CMakeFiles/test1.dir/rule] Error 2
CMakeFiles\Makefile2:146: recipe for target 'test1/CMakeFiles/test1.dir/rule' failed
mingw32-make.exe: *** [test1] Error 2
Makefile:130: recipe for target 'test1' failed
```
3.  `3rd party libraries` not helped. When I paste 3RD_PARTY nothing changed.

Julusian, 2019-05-20:
I just hit this issue, and after some digging I figured out the fix:
https://github.com/ustisha/CMake/compare/master...Julusian:patch-1

ustisha, 2019-05-21:
> I just hit this issue, and after some digging I figured out the fix:
> [ustisha/CMake@master...Julusian:patch-1](https://github.com/ustisha/CMake/compare/master...Julusian:patch-1)

Good Job! 
Dependencies should be linked manually, this is main problem. 
`link_arduino_library(DallasTemperature_Lib OneWire_Lib ${board_id})`

I will try this fix on my project with much libs, but if each lib will be need linked manually it can be problem (( Also i already make some fix in old https://github.com/queezythegreat/arduino-cmake and use it.

closed 2019-05-21 by ustisha

MrPointer, 2019-05-25:
@ustisha Oh, I'm not really familiar with any Arduino libraries so I didn't know **DallasTemperature** depends on **OneWire** 😆 
Yes, that explains it all.
As for the annoyance it might cause, unfortunately there's not much we can do about it, as there's no way for the framework (at least currently or in the near future) to resolve dependencies on its' own.
Moreover, I believe this feature couldn't be implemented in plain CMake, and would require a 3rd-party tool such as **python**.