arduino-cmake/Arduino-CMake-NG #75: Undefined reference to countPulseASM when using pulseIn function

dominicap, 2019-02-26:
OS: macOS
OS Version: 10.14.2
Platform: Arduino
Platform SDK Version: [Arduino SDK version 1.8.8]

When trying to use the built in "pulseIn" function, I receive an undefined reference to `countPulseASM'

Build Output:
```
====================[ Build | auto_car | Default ]==============================
/usr/local/bin/cmake --build /Users/Dominic/Desktop/auto-car/build --target auto_car -- -j 4
[ 92%] Built target mega_atmega2560_core_lib
[ 96%] Linking CXX executable auto_car.elf
/var/folders/vh/n3x031n547v4qhvxs1p_wtf00000gn/T//ccNcSGia.ltrans0.ltrans.o: In function `pulseIn':
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/wiring_pulse.c:46: undefined reference to `countPulseASM'
collect2: error: ld returned 1 exit status
make[3]: *** [auto_car.elf] Error 1
make[2]: *** [CMakeFiles/auto_car.dir/all] Error 2
make[1]: *** [CMakeFiles/auto_car.dir/rule] Error 2
make: *** [auto_car] Error 2
```

CMakeLists.txt:
```Cmake
cmake_minimum_required(VERSION 3.13.3)
project(auto_car LANGUAGES C CXX)

get_board_id(board_id mega atmega2560)

add_arduino_executable(auto_car ${board_id} src/auto_car.cpp)
```

Source file:
```C++
#include <Arduino.h>

#define PIN 14

void setup()
{
    // Setup
}

void loop()
{
    pulseIn(PIN, HIGH, 40000);
}
```

I believe this might be related to [issue 364](https://github.com/sudar/Arduino-Makefile/issues/364) in the Arduino-Makefile community and [issue 23](https://github.com/francoiscampbell/CLionArduinoPlugin/issues/23) from the original Arduino CMake Build System.

Help would be appreciated, thanks!

MrPointer, 2019-02-26:
Quick guess: Try to append `ASM` to the project's languages definition.
Your `project()` call should look like this: `project(auto_car LANGUAGES C CXX ASM)`

dominicap, 2019-02-26:
@MrPointer Yup, that was it, thanks again!

cebas, 2019-04-26:
@MrPointer I had this problem too, and fixed with your workaround. Thanks!
But I think this is still a bug, because the ASM source belongs to the core, not to my project, and it should have been discovered when building the core lib.

MrPointer, 2019-04-27:
@cebas You're probably right - I guess it has something to do with CMake's generator expressions, but I'm not entirely sure.
I'm re-opening this until we have a concrete answer.

dev10, 2020-02-20:
In case someone else is having the same issue with Arduino 1.8.11, I had to make it look like the following:

in CMakeLists.txt line 8:
```
project(${PROJECT_NAME} LANGUAGES C CXX ASM)
```

in cmake/Platform/Arduino.cmake starting line 1477:
```
function(find_sources VAR_NAME LIB_PATH RECURSE)
    set(FILE_SEARCH_LIST
        ${LIB_PATH}/*.cpp
        ${LIB_PATH}/*.c
        ${LIB_PATH}/*.S
...
```

MrPointer, 2020-02-21:
@dev10 I'm sorry but there's no such line in the file `cmake/Platform/Arduino.cmake`. 
In fact, this file only contains 40 lines.

You're probably confusing with the original [arduino-cmake](https://github.com/queezythegreat/arduino-cmake) project, which this project continues since it's the only version of this project that had everything written in a single file (indeed `Arduino.cmake`)

brandentimm, 2020-07-05:
I had the same issue that was fixed by @dev10's suggestion from the Arduino.cmake file generated by the CLion Arduino plugin, just as a general tip/FYI for other users stumbling across this.

allyo-dmytro-smyk, 2021-04-22:
@dev10 that helped a lot, thanks