arduino-cmake/Arduino-CMake-NG #77: Newbie: need some hand holding.

rileyrg, 2019-03-05:
First attempt at using cmake. Errors. My project root CmakeLists.txt:

> cmake_minimum_required(VERSION 3.2)
> project(blink C CXX ASM)
> get_board_id(board_id nano atmega328)
> add_arduino_executable(blink ${board_id} src/main.cpp)
> upload_arduino_target(blink "${board_id}" /dev/ttyACM0)
> 

Q1: where do I find _documentation for get_board_id. I have googled and googled. What should I be passing for an Elegoo Uno R3?

Now to the make process

>  rgr@development-nuc  ~/development/arduino/Blink-PlatformIO/build   master ● ?  cmake -DCMAKE_TOOLCHAIN_FILE=../cmake/Arduino-Toolchain.cmake ..
> -- The C compiler identification is GNU 5.4.0
> -- The CXX compiler identification is GNU 5.4.0
> -- The ASM compiler identification is GNU
> -- Found assembler: /home/rgr/development/arduino/arduinoSDK/hardware/tools/avr/bin/avr-gcc
> -- Arduino SDK version 1.8.8: /home/rgr/development/arduino/arduinoSDK
> -- Determined Platform Header: /home/rgr/development/arduino/arduinoSDK/hardware/arduino/avr/cores/arduino/Arduino.h
> -- Arduino SDK version 1.8.8: /home/rgr/development/arduino/arduinoSDK
> -- Determined Platform Header: /home/rgr/development/arduino/arduinoSDK/hardware/arduino/avr/cores/arduino/Arduino.h
> -- Check for working C compiler: /home/rgr/development/arduino/arduinoSDK/hardware/tools/avr/bin/avr-gcc
> -- Check for working C compiler: /home/rgr/development/arduino/arduinoSDK/hardware/tools/avr/bin/avr-gcc -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Arduino SDK version 1.8.8: /home/rgr/development/arduino/arduinoSDK
> -- Determined Platform Header: /home/rgr/development/arduino/arduinoSDK/hardware/arduino/avr/cores/arduino/Arduino.h
> -- Check for working CXX compiler: /home/rgr/development/arduino/arduinoSDK/hardware/tools/avr/bin/avr-g++
> -- Check for working CXX compiler: /home/rgr/development/arduino/arduinoSDK/hardware/tools/avr/bin/avr-g++ -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> CMake Error at cmake/Platform/Other/TargetFlagsManager.cmake:67 (set_target_compile_flags):
>   set_target_compile_flags Function invoked with incorrect arguments for
>   function named: set_target_compile_flags
> Call Stack (most recent call first):
>   cmake/Platform/Targets/ExecutableTarget.cmake:17 (set_executable_target_flags)
>   CMakeLists.txt:16 (add_arduino_executable)
> 
> 
> CMake Error at CMakeLists.txt:19 (upload_arduino_target):
>   Unknown CMake command "upload_arduino_target".
> 

Thats it. Where do I go from here. I can find nothing at all in google and this CMakesList was lifted from the first page in the Wiki.

Many thanks for helping me out of the quagmire I find myself in! ;)

MrPointer, 2019-03-07:
@rileyrg Hi and welcome to the Arduino-CMake project!
I'm sorry to hear that you're having such a hard time using this framework.

Reviewing your logs, I suspect you've cloned the **master** branch of this repo, which is _**not**_ the stable branch as it is in some other git repos! Instead, here it acts as the main dev branch, while official releases are bundled with [GitHub's Releases](https://github.com/arduino-cmake/Arduino-CMake-NG/releases). Use them (Desirably the latest one).

As for the `board_id` dimness - The function is nowhere to be found on Google since it's defined by the framework and **not** documented anywhere in the Wiki (Not as a standalone function at least).
Generally what it does is constructing a string combining both the ID of the board to be used and possibly its' CPU, that later should be passed as an argument to many other API functions.
Just to make it clear, take for example the `Uno` board you've mentioned - It's not only the name of the board's model, but also it's ID (According to Arduino, not this framework). This is usually like this.
However, looking at the example we see the `nano` board ID and a possibly strange string, `atmega328`. This is the board's CPU, and it's required because some boards are compatible with multiple CPUs.
You shouldn't worry about that, just get some basic info about your board in advance.
Back to the `get_board_id` function - This entire concept you see in the example is about to be deprecated (In fact it's already deprecated in **master**, however not documented yet), omitting the use the `board_id` argument by users like you entirely.
With that being said - Please follow the Wiki docs (And the above explanation) until the new version is officially released.

Now one last thing regarding boards - You've mentioned the `Elegoo Uno R3` board, however, I'm not sure it's entirely supported by the framework yet. It seems to be a 3rd party platform, i.e. not plain, native Arduino, which is currently not supported.
You should first try to use the project after reading what I've written above, but if that doesn't work - Do not be so surprised 😞 

I hope I could en-light your path to happy Arduino-CMake development! 😄

closed 2019-03-15 by MrPointer

rileyrg, 2019-03-15:
Many thanks. I did get it working in the end but not using the documented
samples. I shall get back with my cmake set up.   I had considered using
platformio but I can't see a way to generate compile_commands.json which I
need for the excellent LSP/ccls Emacs integrations. So, thank you. And
you'll hear from me soon.