arduino-cmake/Arduino-CMake-NG #79: RegisterHardwarePlatform doesn't work

PML-Trey, 2019-03-25:
Previous versions of Arduino CMAKE had an API you could use to supply your own distribution of the Arduino embedded software.  It looks like pieces of this have made it into this new version, but the functionality is missing.  It would be nice if this API was added back in.

MrPointer, 2019-03-29:
@PML-Trey Yes it would be nice indeed.  In fact, it's planned to be part of the framework before reaching a stable `1.0` version.
With that being said, there were many problems with the older API and platform-subsystem in general, so it's going to be completely refactored and redesigned, suited to support many different platforms including **ESP**.

davr, 2019-04-03:
Hi, so just to confirm, the current version of Arduino-CMake-NG wont work with any boards except the ones that come with the official arduino sdk, but that support is planned in the near future? 

I'd like to use it with PJRC's Teensy boards, they have really good arduino support generally, I don't know much about the arduino build system, but they install their own boards.txt file in eg arduino-1.8.7/hardware/teensy/avr/boards.txt

MrPointer, 2019-04-04:
@davr Unfortunately you're correct - It's currently not supported, including you specific use case ☹️