arduino-cmake/Arduino-CMake-NG #80: Install for library depending on platform

RobinMaub, 2019-03-29:
Hello,

I would create a library (let's say libArduino) using SD and Servo, and link to this lib in a separated executable, but I get the following error `CMake Error: install(EXPORT "libarduino-targets" ...) includes target "libarduino" which requires target "servoLib" that is not in the export set.`

how can I add to the export set the required files ?

Thanks,

MrPointer, 2019-04-04:
@RobinMaub Hi there and sorry for the delay, could you please attach your **CMakeLists.txt** file?

RobinMaub, 2019-04-05:
@MrPointer Hello, no problem.
Yes sure, attached : 
[CMakeLists.txt](https://github.com/arduino-cmake/Arduino-CMake-NG/files/3046573/CMakeLists.txt)

My goal is to get something like :
1. libarduino (a library) links to Servo or SPI or something like that
2. an arduino executable links to libarduino and is able to use Servo or SPI through it.

MrPointer, 2019-04-05:
Oh I see, it looks like you're using the **master** branch, although it's **_NOT_** designed for users!
For that, you have our [releases](https://github.com/arduino-cmake/Arduino-CMake-NG/releases), where you can find all stable (to a certain degree) versions.
Note that your current **CMakeLists.txt** file will need to change due to this transition, since the current **master** branch introduced many API changes - To understand exactly how the API should be used with the latest release, please see our [docs](https://github.com/arduino-cmake/Arduino-CMake-NG/wiki).

RobinMaub, 2019-04-05:
Ahh My bad...

Can you tell me if the thin I am looking for will be working with my CMakeList ?

Thanks

MrPointer, 2019-04-05:
Yes of course, what your'e trying to achieve is a very common pattern in **Arduino-CMake**.
The only thing I'm not sure of (and haven't been since your original comment) is the `install` part - I don't know what do you intend to use it for, but personally I've never seen any use of it in the **Arduino-CMake** framework.

RobinMaub, 2019-04-05:
Ok many thanks !

closed 2019-04-05 by RobinMaub