arduino-cmake/Arduino-CMake-NG #87: SDK not found on Debian 9 with official Arduino binary

joede, 2019-05-21:
I've tried to build your example "hello-world". My system is a Debian 9 with the official binary tarball installed in `/home/jd/local/arduino-1.8.9`. To test your project, I have clones it into `/home/jd/tmp`. So I select the directory of the example and passed the full path to `Arduino-Toolchain.cmake` and the full path to the *root* of the Arduino SDK path. This is the result:

~~~~
$ cd Arduino-CMake-NG/examples/hello-world
$ mkdir .build && cd .build
$ cmake -DCMAKE_TOOLCHAIN_FILE=../../../cmake/Arduino-Toolchain.cmake -DARDUINO_SDK_PATH=/home/jd/local/arduino-1.8.9 ..
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Arduino SDK version 1.8.9: /home/jd/local/arduino-1.8.9
-- Determined Platform Header: /home/jd/local/arduino-1.8.9/hardware/arduino/avr/cores/arduino/Arduino.h
-- Arduino SDK version 1.8.9: /home/jd/local/arduino-1.8.9
-- Determined Platform Header: /home/jd/local/arduino-1.8.9/hardware/arduino/avr/cores/arduino/Arduino.h
-- Check for working C compiler: /home/jd/local/arduino-1.8.9/hardware/tools/avr/bin/avr-gcc
CMake Error at /home/jd/tmp/Arduino-CMake-NG/cmake/Platform/Other/ArduinoSDKSeeker.cmake:36 (message):
  Couldn't find Arduino SDK path - Is it in a non-standard location?

  If so, please set the ARDUINO_SDK_PATH CMake-Variable
Call Stack (most recent call first):
  /home/jd/tmp/Arduino-CMake-NG/cmake/Arduino-Toolchain.cmake:68 (find_arduino_sdk)
  /home/jd/tmp/Arduino-CMake-NG/examples/hello-world/.build/CMakeFiles/3.13.2/CMakeSystem.cmake:6 (include)
  /home/jd/tmp/Arduino-CMake-NG/examples/hello-world/.build/CMakeFiles/CMakeTmp/CMakeLists.txt:3 (project)

CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error at /usr/share/cmake-3.13/Modules/CMakeTestCCompiler.cmake:37 (try_compile):
  Failed to configure test project build system.
Call Stack (most recent call first):
  CMakeLists.txt:3 (project)

-- Configuring incomplete, errors occurred!
See also "/home/jd/tmp/Arduino-CMake-NG/examples/hello-world/.build/CMakeFiles/CMakeOutput.log".
~~~~

The logfile contains the following:

~~~~
$ cat CMakeOutput.log 
The target system is: Arduino -  - 
The host system is: Linux - 4.9.0-6-amd64 - x86_64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /home/jd/local/arduino-1.8.9/hardware/tools/avr/bin/avr-gcc 
Build flags: 
Id flags:  

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/home/jd/tmp/Arduino-CMake-NG/examples/hello-world/.build/CMakeFiles/3.13.2/CompilerIdC/a.out"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /home/jd/local/arduino-1.8.9/hardware/tools/avr/bin/avr-g++ 
Build flags: 
Id flags:  

The output was:
0

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "a.out"

The CXX compiler identification is GNU, found in "/home/jd/tmp/Arduino-CMake-NG/examples/hello-world/.build/CMakeFiles/3.13.2/CompilerIdCXX/a.out"
~~~~

MrPointer, 2019-05-25:
@joede Hi, please try to set `ARDUINO_SDK_PATH` as an env variable.

joede, 2019-05-28:
Sotry for the delay... I have set `ARDUINO_SDK_PATH` in my `~/.profile` and removed it from the parameters list of my CMake call. And it works! Thank you!

closed 2019-05-28 by MrPointer