arduino-cmake/Arduino-CMake-NG #92: find_arduino_library simply doesn't work

gnthibault, 2019-06-21:
I am trying to compile a slightly modified version of StandardFirmata arduino example sketch for arduino uno
Here is my CMakeLists.txt:

```
# Define CMake's minimum version (must-do) and the project's name and supported languages
cmake_minimum_required(VERSION 3.8.2)
project(Firmata)

# Call a framework utility function, passing it information about the hardware board that will be used - This function returns a structure known only to the framework
get_board_id(board_id uno atmega328p)

find_arduino_library(servo_lib Servo ${board_id})
find_arduino_library(firmata_lib Firmata ${board_id})

# Create an executable suitable for Arduino using CMake-style target-creation
add_arduino_executable(Firmata ${board_id} Firmata.cpp)
# Upload the created target through a connected Serial Port (Where your board is connected to)
upload_arduino_target(Firmata "${board_id}" /dev/ttyACM0)
#You should then call CMake (either through cmd, cmake-gui or an IDE if it supports that), passing it the argument -DCMAKE_TOOLCHAIN_FILE=[project_path]/cmake/Arduino-Toolcha

# Now link libraries
link_arduino_library(Firmata servo_lib ${board_id})
link_arduino_library(Firmata firmata_lib ${board_id})

```

The error is the following:

> CMake Error at /home/user/projects/Arduinotest/Platform/Targets/ArduinoCMakeLibraryTarget.cmake:63 (get_target_property):
>   get_target_property() called with non-existent target "uno_core_lib".

gnthibault, 2019-06-21:
Edit: Actually, fix for this issue was mentionned by @MrPointer  in https://github.com/arduino-cmake/Arduino-CMake-NG/issues/55

It is a matter of order

MrPointer, 2019-06-21:
@gnthibault Hi there - What branch are you using? Is it a release?

gnthibault, 2019-06-21:
@MrPointer  I did used a release, in the changelog it is written: ## Version 0.7

MrPointer, 2019-06-22:
@gnthibault Well that might explain it, as this bug is possibly already solved in the dev branch, which is `master` in our case. 
You can try and clone it to see if that solves your issue, but do note - The API has changed (quite significantly actually) and there's not much docs about it yet (As it hasn't been officially released). 
With that being said, I will gladly lead your way through this process, as I still don't know when the new version is gonna come out.

ZsoltSaskovy, 2019-07-14:
@MrPointer I might be wrong, but I missed the functionality of "find_arduino_library" to add the directories of the header files to the include_directories list. Is it only me who can't use the framework properly, or there is no such functionality? I implemented it on my own. Would it make a sense to create a pull request of it, or it's less needed by others? Thanks, Zsolt

MrPointer, 2019-07-17:
@ZsoltSaskovy I'm afraid I don't quite understand your precise intention.
Please do open a PR, it'll be better to discuss it there.