arduino-cmake/Arduino-CMake-NG #93: ARDUINO_SDK_PATH variable is forgotten during compiler check (v0.6.1)

lucaspcamargo, 2019-09-25:
This also happened using the current master, but I decided to go back to the latest release to ensure that it was not because master was in a WIP state.

I am trying to configure a simple project, but it fails as such:
```
$ cmake -DARDUINO_SDK_PATH=/home/camargo/sw/arduino-1.8.9 -DCMAKE_TOOLCHAIN_FILE=/home/camargo/git/ghc-fw/Arduino-CMake-NG/cmake/Arduino-Toolchain.cmake ..
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/camargo/sw/arduino-1.8.9/hardware/tools/avr/bin/avr-gcc
-- Arduino SDK version 1.8.9: /home/camargo/sw/arduino-1.8.9
-- Determined Platform Header: /home/camargo/sw/arduino-1.8.9/hardware/arduino/avr/cores/arduino/Arduino.h
-- Arduino SDK version 1.8.9: /home/camargo/sw/arduino-1.8.9
-- Determined Platform Header: /home/camargo/sw/arduino-1.8.9/hardware/arduino/avr/cores/arduino/Arduino.h
-- Check for working C compiler: /home/camargo/sw/arduino-1.8.9/hardware/tools/avr/bin/avr-gcc
CMake Error at /home/camargo/git/ghc-fw/Arduino-CMake-NG/cmake/Platform/Other/ArduinoSDKSeeker.cmake:36 (message):
  Couldn't find Arduino SDK path - Is it in a non-standard location?

  If so, please set the ARDUINO_SDK_PATH CMake-Variable
Call Stack (most recent call first):
  /home/camargo/git/ghc-fw/Arduino-CMake-NG/cmake/Arduino-Toolchain.cmake:66 (find_arduino_sdk)
  /home/camargo/git/ghc-fw/build/CMakeFiles/3.10.2/CMakeSystem.cmake:6 (include)
  /home/camargo/git/ghc-fw/build/CMakeFiles/CMakeTmp/CMakeLists.txt:3 (project)

CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error at /usr/share/cmake-3.10/Modules/CMakeTestCCompiler.cmake:37 (try_compile):
  Failed to configure test project build system.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

-- Configuring incomplete, errors occurred!
See also "/home/camargo/git/ghc-fw/build/CMakeFiles/CMakeOutput.log".
!
```

So, the build system seems to understand where my SDK is (`/home/camargo/sw/arduino-1.8.9`),  but ArduinoSDKSeeker.cmake seems to have not gotten the value. My cmake version is 3.10.2. Any help is appreciated.

MrPointer, 2019-09-29:
@lucaspcamargo As stated [here](https://github.com/arduino-cmake/Arduino-CMake-NG/wiki/Installation#arduino-sdk-path), you should set the `ARDUINO_SDK_PATH` variable as an environment variable. Passing it as a CMake argument doesn't work, as far as I know.

lucaspcamargo, 2019-10-02:
Oh, sorry. That sailed right over my head. Everything works as expected now.

closed 2019-10-02 by lucaspcamargo