arduino-cmake/Arduino-CMake-NG #94: Support for Third-Party Boards - Game plan

jdalrymple, 2019-11-04:
Hey!

I've looked through most of the related issues in this repo (#34)  and the archived repo as well as some PR's to get some background. That being said, with this newer version of the library, how do you see the implementation for supporting third party boards? Did you have a master plan that the community could start working on besides the two issue road map ?

MrPointer, 2019-11-04:
@jdalrymple Hi, sorry to disappoint but I no longer actively maintain this repo ☹️ 
I've been maintaining this project longer than planned - I don't even use Arduino nor anything-related in my work (both personal and career-wise). I did it mostly to help the community at the time, as I realized how hard it was to develop on anything other than the Arduino IDE, following a one-time experience I had with Arduino.

Moreover, I honestly believe that the current CMake approach to Arduino, in general, is not the correct one, especially as the [Arduino-CLI](https://github.com/arduino/arduino-cli) is being seriously developed now. It seems to be an interesting project that could be wrapped in many ways other than CMake, and as IDEs such as CLion are implementing support for many build systems, practically anything could be used.
I guess that I mainly support this mindset due to difficulties I had trying to implement some advanced features with CMake, such as the one you were referring to, only to realize it's not worth it, if even possible. CMake is a great tool and language, but there is only so much that it can do, and I think we're close to its' limit here in this project.

jdalrymple, 2019-11-04:
Makes sense! I've been looking for a solution to the restrictions brought on by the IDE, specifically the folder structure. Regardless, i appreciate your work on this repo! Maybe ill switch my build system after looking at the cli :D

jdalrymple, 2019-11-05:
You can close this if you'd like! I kept it open since you pinned it.

MrPointer, 2019-11-06:
Yeah I know, it just seems important for other people to know as well, at least until I'll post it officially.